mazupo/pi-webrtc #82: Lookinf for the right QEMU env for compiling the project

eliabruni, 2023-11-10:
Since the compilation of all libraries are quite heavy, let's try to find the right QEMU setting to compile them all on an x86_64 machine. These are my cross compilation instructions that so far managed to compile `libwebrtc.a`. However, I encountered issues when trying to compile `microsoft-signalr.so` at the `casablanca` step (see also [here](https://github.com/TzuHuanTai/RaspberryPi_WebRTC/issues/79#issuecomment-1801378957)).

Let me first give the specs of the system I used

```
# Operating System and Version:
lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.5 LTS
Release:	20.04
Codename:	focal

# kernel
uname -r
5.15.0-88-generic

# architecture
uname -m
x86_64
```

1. **ARM64 Root Filesystem Setup**:
```bash
# Install the ARM64 cross-compilation toolchain and other necessary packages
sudo apt-get update
sudo apt-get install gcc-aarch64-linux-gnu g++-aarch64-linux-gnu binutils-aarch64-linux-gnu qemu-user-static debootstrap
sudo debootstrap --arch arm64 --foreign bullseye rootfs
sudo cp /usr/bin/qemu-aarch64-static rootfs/usr/bin/
```

2. **Chroot Environment Setup**:
```bash
sudo mount -o bind /dev rootfs/dev
sudo mount -t devpts -o gid=5,mode=620 devpts rootfs/dev/pts
sudo chroot rootfs /bin/bash
mount -t proc /proc /proc
```

3. **Debootstrap Initialization**:
```bash
/debootstrap/debootstrap --second-stage
```

4. **Basic Package Installation**:
```bash
apt-get update
apt-get install -y git lsb-release sudo curl
```

5. **User Environment Setup**:
```bash
useradd -m pi
echo "pi ALL=(ALL) NOPASSWD:ALL" | tee -a /etc/sudoers
su - pi
bash
```

6. **Additional Dependencies Installation**:
```bash
sudo apt-get update
sudo apt-get install -y g++ libasound2-dev libpulse-dev libudev-dev libexpat1-dev libnss3-dev libgtk-3-dev python pulseaudio libpulse-dev build-essential libncurses5 libx11-dev
sudo apt remove -y libssl-dev
pulseaudio --start
```

7. **Depot Tools Setup**:
```bash
cd /home/pi
git clone https://chromium.googlesource.com/chromium/tools/depot_tools.git
echo "export PATH=$(pwd)/depot_tools:\$PATH" >> ~/.bashrc
source ~/.bashrc
```

8. **Sysroot Dependencies for ARM64**:
```bash
sudo apt-get install -y gcc-aarch64-linux-gnu g++-aarch64-linux-gnu binutils-aarch64-linux-gnu
```

9. **WebRTC Source Fetching**:
```bash
mkdir -p ~/webrtc && cd ~/webrtc
fetch --nohooks webrtc
src/build/linux/sysroot_scripts/install-sysroot.py --arch=arm64
gclient sync -D
cd src
git checkout -b local-5790 branch-heads/5790
git gc --aggressive
gclient sync -D --force --reset --with_branch_heads --no-history
```

10. **Clang Setup**:
```bash
cd /home/pi
curl -OL https://github.com/llvm/llvm-project/releases/download/llvmorg-16.0.6/clang+llvm-16.0.6-aarch64-linux-gnu.tar.xz
tar Jxvf clang+llvm-16.0.6-aarch64-linux-gnu.tar.xz
echo 'export PATH=/home/pi/clang+llvm-16.0.6-aarch64-linux-gnu/bin:$PATH' >> ~/.bashrc
source ~/.bashrc
```

11. **Ninja Building and Installation**:
```bash
git clone https://github.com/martine/ninja.git
cd ninja
./configure.py --bootstrap
mv /home/pi/depot_tools/ninja /home/pi/depot_tools/ninja_org
cp /home/pi/ninja/ninja /home/pi/depot_tools/ninja
```

12. **GN Building and Installation**:
```bash
cd /home/pi
git clone https://gn.googlesource.com/gn
cd gn
sed -i -e "s/-Wl,--icf=all//" build/gen.py
python build/gen.py
ninja -C out
sudo mv /home/pi/webrtc/src/buildtools/linux64/gn /home/pi/webrtc/src/buildtools/linux64/gn_org
cp /home/pi/gn/out/gn /home/pi/webrtc/src/buildtools/linux64/gn
sudo mv /home/pi/depot_tools/gn /home/pi/depot_tools/gn_org
cp /home/pi/gn/out/gn /home/pi/depot_tools/gn
```

13. **Extra Dependencies for GN**:
```bash
sudo apt-get update
sudo apt-get install python3-pip -y
pip3 install setuptools
```

14. **WebRTC Build Configuration**:
```bash
gn gen out/Release64 --args='target_os="linux" target_cpu="arm64" rtc_include_tests=false rtc_use_h264=false use_rtti=true is_component_build=false is_debug=false rtc_build_examples=false use_custom_libcxx=false rtc_build_tools=false rtc_use_pipewire=false clang_base_path="/home/pi/clang+llvm-16.0.6-aarch64-linux-gnu" clang_use_chrome_plugins=false'
ninja -C out/Release64
```

15. **Header Files Synchronization**:
```bash
sudo apt-get install rsync
rsync -amv --include=*/ --include=*.h --include=*.hpp --exclude=* ./ ./include
```

16. **Compiled libwebrtc.a Location**:
```bash
/home/pi/webrtc/src/out/Release64/obj/libwebrtc.a
```

closed 2023-11-13 by TzuHuanTai

TzuHuanTai, 2023-11-13:
I tested and updated the steps for all compilations.

eliabruni, 2023-11-14:
Awesome!