mazupo/pi-webrtc #178: Can I use a web browser instead of the Pi Camera app?

samthelion17, 2024-09-18:
This would allow me to view the stream from a bigger screen, like a laptop for example. Just the other day I was testing mediamtx for a project like this, so far I'm liking your implementation better.

TzuHuanTai, 2024-09-18:
Yes, you can! I had a web-based UI actually, but I hadn't finished it yet, and then I switched to app implementation a few months ago.

samthelion17, 2024-09-18:
How? Shed some light plz 🙏

TzuHuanTai, 2024-09-18:
I used mqtt.js and reused the code from the webrtc official javascript demo from their [Github](https://github.com/webrtc/samples), then, changed the signaling mechanism from the sample code to yours.
Although it doesn't seem too difficult, there's a lot of extra work. :(

KevinXuxuxu, 2024-09-23:
Could you opensource (or share) the code for the web UI version? This would enable accessing the video streaming from literally any device (instead of just android phone).

TzuHuanTai, 2024-09-23:
I do have a plan to release a web UI that everyone can use like the Pi Camera app. Other features such as event trigger, motion detection, or p2p control gpio. However, I’m developing purely out of personal passion and ideals, without any funding or sponsorship. I’m only able to work on it during my spare time, the progress might be slower than expected. I hope this answers your question.

obenchekroun, 2024-10-10:
Hi !

Really great project ! It works really well and I am impressed by the minimal delay it has !
Adding to @samthelion17, it would be great if you release the crude web demo you used, for testing with the web interface !

Thank you very much

TzuHuanTai, 2024-10-11:
First of all, I appreciate you guys like this project.
I just finished the setting page in the past few days, the next step is to migrate the webrtc streaming code.
If all goes well, I expect the first Web UI to be released by the end of this month.

[image]
[image]

obenchekroun, 2024-10-11:
that's great ! thanks a lot for all this work !

TzuHuanTai, 2024-10-13:
I have a question. Could you guys share some reasons/purposes why you need a Web UI? 
Knowing the requirements helps me design the Web UI, I hope it's easy to use for all people, not just me.

samthelion17, 2024-10-13:
> I have a question. Could you guys share some reasons/purposes why you need a Web UI? Knowing the requirements helps me design the Web UI, I hope it's easy to use for all people, not just me.

For me, I just want to be able to view the live feed on a bigger display and on any device, no matter the OS.

TzuHuanTai, 2024-10-16:
Thank all the replies.
I'm just curious what people are looking for. It might change the priority of the new feature.

Answer @beyonlo 's questions.
I used to develop it in Angular with the PWA feature but moved to the React-based framework this year. Now, I’m using React for the web and React Native for the Android app.

By the way, the live-streaming page is finished.
[image]

obenchekroun, 2024-10-16:
Hi @TzuHuanTai,

It looks great !

> I have a question. Could you guys share some reasons/purposes why you need a Web UI? Knowing the requirements helps me design the Web UI, I hope it's easy to use for all people, not just me.

For me, I have a few reasons :
- My primary phone is an iPhone so being usable on any OS is particularly useful
- I often work from home, and use it to keep an eye on my baby sleeping in his crib. Being able to put it in a corner window of my screen would be much appreciated, instead of having the app open on a phone

Hope this helps !

silent817, 2024-10-19:
I see this as great project, I'll buy zero2W to try it.
Only one question about webUI: can live view be opened just with one link? I am interested in adding it as start page to the browser on my old device and then opening the browser will start streaming automatically. Would be great doorbell camera I believe.

TzuHuanTai, 2024-10-25:
Here is the web ui: [Pi Camera Web](https://picamera.live/)
All setup process are the same as what you do on the app.

TzuHuanTai, 2024-10-25:
The ui source code is not provided.

obenchekroun, 2024-10-25:
Hi @TzuHuanTai !

Thanks for the work ! 
I have spent a bit of time testing it. It works really well on desktop (macOS Sonoma, Safari version 17.5), and the configuration is , but the sound doesn't work on my iPhone (iPhone 14 pro, iOS 16.5, with Safari as a web browser). Everything else works on phone (preview, video) but not the phone.

Please feel free to ask any question, I will try to help as much as I can.

> The ui source code is not provided.
On another note, are planning on releasing the source code,

Thank you very much

TzuHuanTai, 2024-11-02:
Sorry for the late reply.
I'm thinking whether make a javascript package this few days.
If I write a c++ the client side, and package it up by python, js, java, c#, or so. That means all coding language can easily use it.

In fact, my client source code has hundreds or thousands of lines, which use different module, monitor connection status, and control a data transfer via datachannel.
I told with some of my friends, they said even though webrtc official site have a lot of example, it might take a lot of time just to learn it, and the code only handle the webrtc connection needs about 100 rows. As a frontend engineer, most of people don't have video streaming background, if there are some third-party package like [videojs-webrtc-plugin](https://github.com/ant-media/videojs-webrtc-plugin), it would be much friendly and help everyone across the WebRTC berrier.

I assume the api look like this indeed simplify a lot of code:
```js
let videoElement = document.getElementById('videoElement');
const congif = {
   rpiUid: xxxx,
   mqttHost: xxxx,
   mqttPort: xxxx,
   ....
};
let connObj = new PiCamera(congif);
connObj.attachMediaElement(videoElement);
connObj.connect();
connObj.snapshot();
connObj.onSnapshot((data:Uint8Array)=>{
    ...
});
connObj.disconnect();
```
This is the best way I've found so far, but I haven't made the decision to develop a package yet. There's still a lot of things need to take into account.

TzuHuanTai, 2024-11-08:
I've tested the javascript client package for connecting to `pi_webrtc`.
npm: https://www.npmjs.com/package/picamera.js
github: https://github.com/TzuHuanTai/PiCamera.js

closed 2025-04-17 by TzuHuanTai