mazupo/pi-webrtc #179: RPi 5 Compatibility

Harvey-Mushman, 2024-09-18:
Your comment about RPi5 not supporting v4l2 hardware encoding and suggest there is a solution for software encoding but stop short of going into any detail how this might be done.  Can you report this project where you start out with a RPi5 or describe how and where to do the encoding in software?  This is a great project that has so many possibilities but the hardware encoding restriction seems like a big hiccup.

TzuHuanTai, 2024-09-18:
Sorry that my document only writes one line to describe [here](https://github.com/TzuHuanTai/RaspberryPi_WebRTC/blob/fc99cf384e2b21f0b432f644ed3b155c344545b3/README.md?plain=1#L75)
To use software encoding, please set `v4l2_format` value to `mjpeg` and remove `--hw_accel` flag.

Below is a command used for software encoding and set `uuid=home-pi-5` for example.
```bash
/path/to/pi_webrtc 
--device=/dev/video0 \ # camera device path
--fps=30 \
--width=1280 \
--height=960 \
--v4l2_format=mjpeg \ #use mjpeg here and remove --hw_accel flag
--mqtt_host=example.s1.eu.hivemq.cloud \ # set you mqtt url
--mqtt_port=8883 \
--mqtt_username=hakunamatata \ # set you mqtt username
--mqtt_password=Wonderful \ # set you mqtt pw
--uid=home-pi-5 # set your custom uid
--record_path=/mnt/ext_disk/video/ # remove this line if recording is not requred
```

The following MQTT topics will be used, `{mqtt_user_id_from_app}` is a random value for each new MQTT connection. `home-pi-5` will be changed to your `uid` setting :
1. The user's app publishes `SDP/ICE` to topics below, subscribe by pi_camera
`home-pi-5/ice/{mqtt_user_id_from_app}/offer`
`home-pi-5/sdp/{mqtt_user_id_from_app}/offer` 
2. The `pi_camera` responds `SDP/ICE` to topics below, subscribe by the user's app
`home-pi-5/ice/{mqtt_user_id_from_app}`
`home-pi-5/sdp/{mqtt_user_id_from_app}`

[image]

TzuHuanTai, 2024-09-19:
I'll write all the details on the project's [wiki page](https://github.com/TzuHuanTai/RaspberryPi_WebRTC/wiki) in the next few days.

Harvey-Mushman, 2024-09-19:
Hi Tzu,

 

I really appreciate the response.  If you can help me get this figured out, I will be happy to donate dollars to your efforts.  The WiKi docs are already looking very promising.

 

Best,

--Harvey

Harvey-Mushman, 2024-09-23:
Hi Tzu,

 

I followed the video and github information and I think everything is working… but I don’t know how to start the app.  I guess I’m dumb but I could not find the answer in your YouTube, Github or Wiki pages. 

 

Can you help?

 

Best,

Harvey

TzuHuanTai, 2024-09-23:
1. Download the app [Pi Camera](https://play.google.com/store/apps/details?id=com.tzu.huan.tai.picamera) from Play Store.
2. Open the app go to `Settings` page, and click the **earth icon** to input your MQTT server. Leave `STUN` and `TURN` blank.
3. Press **+** button to create a new device, copy the `uid` from the app, and run `pi_webrtc` with this uid, such as `pi_webrtc --device=/dev/video0 --uid=xxxxxxxxxx ......`

If all is set, you can see the preview on the Pi Camera home page.
You can follow the [readme](https://github.com/TzuHuanTai/Pi-Camera) in the Pi Camera instruction repo for more details.

Harvey-Mushman, 2024-09-24:
Hi Tzu,

 

I followed the directions below and got the app on my android phone to run.  When I try to start pi_webrtc I’m getting an error (see below).  My RPi is model 5 and I removed the hardware encoding and changed the v4l2 format to be mpjeg.  But have no clue why the device can’t be found.  Any suggestions?

 

/home/metwo/pi_webrtc --device=/dev/video0 --fps=30 --width=1280 --height=960 --v4l2_format=mjpeg --mqtt_host=b4ce6ff6fae24b1bbfaa988f8b0c0cfb.s1.eu.hivemq.cloud --mqtt_port=8883 --mqtt_username=veniceMaps --mqtt_password=Columbus1 --uid=d494fa90-d54a-4c28-abc5-3f12c1565faa
Use mjpeg format source in v4l2
[v4l2_utils] Error: v4l2 open(/dev/video0): No such file or directory

 

Best,

Harvey

Harvey-Mushman, 2024-09-24:
Hi again…

 

I have done some testing and don’t see where video0 is available on my RPi5.

 

***@***.***:~ $ v4l2-ctl --list-devices
pispbe (platform:1000880000.pisp_be):
/dev/video20
/dev/video21
/dev/video22
/dev/video23
/dev/video24
/dev/video25
/dev/video26
/dev/video27
/dev/video28
/dev/video29
/dev/video30
/dev/video31
/dev/video32
/dev/video33
/dev/video34
/dev/video35
/dev/video36
/dev/video37
/dev/media0
/dev/media1

rpivid (platform:rpivid):
/dev/video19
/dev/media2

Cannot open device /dev/video0, exiting.
***@***.***:~ $ 

 

 

I tried video19, video20 and media0 and got error messages about fps(30): Inappropriate loctl for device.  So I tried changing the fps 20 15 and still got the same error.

 

By the way, my direct email address is: ***@***.***

 

Best,

Harvey

TzuHuanTai, 2024-09-24:
`/dev/video0` not found, you can follow the issue (#182), the main reason is that the `gpu_mem=16` is not enough for Pi 4B.
You can try `gpu_mem=256` in `/boot/firmware/config.txt` for your Pi 5.

Harvey-Mushman, 2024-10-07:
Hi Tzu,

 

How is it going, wondering if you received the RPi5 yet and if you have had a chance to look at the problems?

 

Best,

--Harvey

TzuHuanTai, 2024-10-07:
Hi Harvey,

Pi 5 will arrive tomorrow. I've done libcamera api to read the camera on pi
os and tested for 3 days already on my Pi 4B to ensure the stability. Once
Pi 5 arrive, I can insert the sd card into it immediately to verify if it
works properly or not.

By the way, I started developing a new web UI few days ago. Just so you
know it's under construction.🙂

Harvey-Mushman, 2024-10-07:
Hi Tzu,

 

You are the expert but I’m not sure when you create the SD card with RPi4 vs. RPi5 if it uses the same OS image.  When you run Imager there is a selection for RPi5 now.

 

Looking forward to new version.

 

Best,

--Harvey

TzuHuanTai, 2024-10-10:
I received pi 5 today! It works fine with the `--use_libcamera` flag on the latest `v1.0.1` pi_camera.
Since pi 5 does not support a hardware encoder, I expected it would cost a lot of CPU usage, but it only uses 25% at 1080p30!
The performance surprises me, and so does the temperature, which goes up to 85°C if there isn't a heat sink...

```
/path/to/pi_webrtc --device=/dev/video0 --fps=30 --width=1920 --height=1080  --mqtt_host=xxxx --mqtt_port=8883 --mqtt_username=xxxx --mqtt_password=xxxx --uid=home-pi-5 --record_path=/mnt/ext_disk/video/ --use_libcamera
```

Harvey-Mushman, 2024-10-10:
Hi Tzu,

 

Wow, that was quick fix…  Congratulations!

 

The heat seems to be an issue of the higher performance CPU.  I have been running mine without a heat sink for over a month now and not noticed any problems.  Before I deploy I will add the recommended heat sink and a fan but in the meantime I am doing this burn-in test.

 

I appreciate your help.

 

Best,

Harvey

TzuHuanTai, 2024-10-11:
I found an issue with the camera module 3.
The official doc said it supports autofocus but it does not work here.
Maybe I need to control it by `libcamera` actively.

Harvey-Mushman, 2024-10-12:
Hi Tzu,

The issuers you are having with Autofocus might be explained in the manual I found at the following address:

https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf

Let me know if this helps, I really do not understand what the camera interface is describing or what problems you are having but trying to learn more about the RPi.

BTW: is there a way to not use the GitHub email address and have our conversations without them in the middle?

Best,
Harvey

TzuHuanTai, 2024-10-13:
Yes, the doc said the autofocus needs to be set in code while the camera starts running.

You could send messages to my email, LinkedIn, or the [Google Group](https://groups.google.com/g/pi-camera). Or is there any other way you like?

TzuHuanTai, 2024-10-13:
Auto focus is fixed in #194

Harvey-Mushman, 2024-10-14:
Hi Tzu,

 

Today I started back on trying to get the app to run.  However, for some reason the RPi program won’t start.  I have attached a screen shot where you can see the app was downloaded and extracted to the root user folder.  When I enter the pi_webrtc command, I get and error message telling me “command not found”.  Yet it is there and before in your prior version it did run.  Any suggestions, I feel bad because this seems like a dumb mistake on my part!

 

Best,

Harvey

Harvey-Mushman, 2024-10-15:
Hi Tzu,

I realized this morning after waking up that I had uncommented the camera in the /boot/firmware/config.txt file to test if the camera was working correctly.  Now your settings are enabled and the RPi was rebooted.  This is causing a new error which reads: 

[v4l2_utils] Error: v4l2 open(/dev/video0): No such file or directory

Is there a particular file folder that is required with I installed your programs?  I'm not clear on where the application needs to be installed into and when I uncompressed, the routine asked if the parent path was to be preserved.  I assumed no but perhaps this is a mistake.

Any suggestions?

--Harvey

TzuHuanTai, 2024-10-15:
It can be placed in any folder where you can execute it. This error represents the camera is not detected by the OS.
What is your setting in `/boot/firmware/config.txt` right now? 
What version of Pi OS are you using? (you can use `uname -a` in terminal)
What is the command you ran `pi_camera`?

closed 2025-04-17 by TzuHuanTai