mazupo/pi-webrtc #180: Execution error:

fabiox23, 2024-09-19:
Running on a RPi 2W with a imx219 camera. The config.txt has been updated as per instructions.

`> ./pi_webrtc --device=/dev/video0 --fps=30 --width=1280 --height=960 --v4l2_format=h264 --hw_accel`

produces the following error:

```
Use h264 format source in v4l2

#
# Fatal error in: ../../media/engine/adm_helpers.cc, line 39
# last system error: 88
# Check failed: 0 == adm->Init() (0 vs. -1)
# Failed to initialize the ADM.Aborted
```

Not using pulseaudio as I have no microphone. What am I doing wrong?

Thanks a lot.

TzuHuanTai, 2024-09-19:
It'll read the "silence" voice even though there isn't a microphone, the `PulseAudio` is always required!
https://github.com/TzuHuanTai/RaspberryPi_WebRTC/blob/d226407391e0530598433c6e62406928d73c4bed/src/conductor.cpp#L275-L276

TzuHuanTai, 2024-10-25:
The "--no_audio" flag is supported starting from the "v1.0.2" release.
It can run `pi_camera` without pulseaudio or any audio source.

```bash
./pi_webrtc --device=/dev/video0 --fps=30 --width=1280 --height=960 --v4l2_format=h264 --hw_accel --no_audio
```

closed 2025-04-17 by TzuHuanTai