mazupo/pi-webrtc #182: /dev/video0 not found

KevinXuxuxu, 2024-09-21:
I'm using a rpi 4 (not zero 2w) but I assume it should work as well?

Have the rpi camera v2 plugged in and verified that it works with the traditional `rpicam-still` and `rpicam-vid` tools.

I was following the [setup steps](https://github.com/TzuHuanTai/RaspberryPi_WebRTC?tab=readme-ov-file#environment-setup) but after I add the lines in `/boot/firmware/config.txt` and restarted rpi, I cannot find the `/dev/video0` file pointer. There is only a `/dev/video19`, which doesn't work when used with `pi_webrtc`:
```
$ ./pi_webrtc --device=/dev/video19 --fps=30 --width=1280 --height=960 --v4l2_form
at=h264 --hw_accel --mqtt_host=... --mqtt_port=8
883 --mqtt_username=... --mqtt_password=... --uid=home-pi-zero2w
Use h264 format source in v4l2
[v4l2_utils] Error: fd(26) set fps(30): Inappropriate ioctl for device
```
Using `v4l2-ctl` to check device:
```
$ v4l2-ctl --list-devices
rpivid (platform:rpivid):
        /dev/video19
        /dev/media0

Cannot open device /dev/video0, exiting.
```

The following is the `/boot/firmware/config.txt` file content for reference:
```
# For more options and information see
# http://rptl.io/configtxt
# Some settings may impact device functionality. See link above for details

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Additional overlays and parameters are documented
# /boot/firmware/overlays/README

# Automatically load overlays for detected cameras

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Automatically load initramfs files, if found
auto_initramfs=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Don't have the firmware create an initial video= setting in cmdline.txt.
# Use the kernel's default instead.
disable_fw_kms_setup=1

# Run in 64-bit mode
arm_64bit=1

# Disable compensation for displays with overscan
disable_overscan=1

# Run as fast as firmware / board allows
arm_boost=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[cm5]
dtoverlay=dwc2,dr_mode=host

[all]

camera_auto_detect=0
start_x=1
gpu_mem=16
```

I image that this file should be part of the rpi OS Lite image so there's shouldn't be any problem with that?
Any help is appreciated, thanks!

TzuHuanTai, 2024-09-21:
My Pi 4B used to run on Bullseye(6.1.21) perfectly so far. I just pulled out the Bookworm lite OS (6.6.31) sd card from Pi Zero 2W and inserted it into my Pi 4b. I found the same `/boot/firmware/config.txt` setting led to different results on Pi 4B!😕

Please increase GPU memory to 256MB in the `config.txt` and reboot. I guess Pi 4B hardware consumes more resources than Pi Zero, hence 16MB is not enough to start hardwares. 
```ini
# Automatically load overlays for detected cameras
#camera_auto_detect=1
camera_auto_detect=0
start_x=1
gpu_mem=256
```
The results of Pi 4B are like the following:
```ini
$ vcgencmd get_camera
supported=1 detected=1, libcamera interfaces=0

$ v4l2-ctl --list-devices
bcm2835-codec-decode (platform:bcm2835-codec):
        /dev/video10
        /dev/video11
        /dev/video12
        /dev/video18
        /dev/video31
        /dev/media2

bcm2835-isp (platform:bcm2835-isp):
        /dev/video13
        /dev/video14
        /dev/video15
        /dev/video16
        /dev/video20
        /dev/video21
        /dev/video22
        /dev/video23
        /dev/media1
        /dev/media3

mmal service 16.1 (platform:bcm2835_v4l2-0):
        /dev/video0

rpivid (platform:rpivid):
        /dev/video19
        /dev/media0

$ v4l2-ctl -d /dev/video0 --list-formats
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'YU12' (Planar YUV 4:2:0)
        [1]: 'YUYV' (YUYV 4:2:2)
        [2]: 'RGB3' (24-bit RGB 8-8-8)
        [3]: 'JPEG' (JFIF JPEG, compressed)
        [4]: 'H264' (H.264, compressed)
        [5]: 'MJPG' (Motion-JPEG, compressed)
        [6]: 'YVYU' (YVYU 4:2:2)
        [7]: 'VYUY' (VYUY 4:2:2)
        [8]: 'UYVY' (UYVY 4:2:2)
        [9]: 'NV12' (Y/UV 4:2:0)
        [10]: 'BGR3' (24-bit BGR 8-8-8)
        [11]: 'YV12' (Planar YVU 4:2:0)
        [12]: 'NV21' (Y/VU 4:2:0)
        [13]: 'RX24' (32-bit XBGR 8-8-8-8)
```

KevinXuxuxu, 2024-09-21:
Thank you for the prompt response! I wouldn't think of the gpu_mem to be the issue without your help.
This solution indeed works and I'm able to get live video end-to-end now!
I think it's worth mentioning in the README about this so I created a pull request https://github.com/TzuHuanTai/RaspberryPi_WebRTC/pull/184

closed 2024-09-22 by TzuHuanTai