mazupo/pi-webrtc #183: I cannot get it to work with my self-hosted MQTT broker (mosquitto)

samthelion17, 2024-09-21:
A "using your self-hosted MQTT" section in the README would be highly appreciated.

TzuHuanTai, 2024-09-22:
You can follow the setup the mosquito broker in [/doc/SETUP_MOSQUITTO.md](https://github.com/TzuHuanTai/RaspberryPi_WebRTC/blob/main/doc/SETUP_MOSQUITTO.md).

xthree, 2024-09-27:
I have followed the directions in the doc, but when I connect from the app, I get this error found in `/var/log/mosquitto/mosquitto.log`

```
1727413896: Client <unknown> disconnected due to protocol error.
1727413896: New connection from 192.168.3.126:48792 on port 1883.
```

Any idea what might cause this?

TzuHuanTai, 2024-09-27:
I'm not sure if the reason is `protocol mqtt` missed in config, because I have never encountered this situation.
You can refer to [this](https://github.com/mqttjs/MQTT.js/issues/1021) post, then try to connect mosquitto from mqtt client tools such as [mqttx](https://mqttx.app/) to make sure all protocol is working properly.

xthree, 2024-09-27:
I tried adding `protocol mqtt`. Still doesn't work.

I can connect successfully with MQTTX with the following settings.

[image]

[image]

I have tried blanking out the `Path` in the app, and still it does not work.

[image]

Are there any certificates for SSL that the app might be trying to apply to the connection by chance?

Thanks for the reply

xthree, 2024-09-27:
I watched your tutorial video and realized I needed to put the websocket port (8083, NOT 1883) in the app. I did that, but it does not work, and in the `/var/log/mosquitto/mosquitto.log` there is no record of a connect attempt. No error, no success.

I am able to connect using MQTTX with Websocket (8083) and MQTT (1883) protocols

[image]

TzuHuanTai, 2024-09-27:
I only use my local mosquito without SSL/TLS in development.
The app only accepts the mqtt connection over SSL/TLS. Or we can say all app listings on the Play Store require an encrypted connection. If not, I have to create a URL whitelist in the app config before compiling, and Google would show alarms if my app accepts unencrypted connections.
I recommend you apply SSL/TLS to your mosquito, even if it's local. I know it's inconvenient. :(

The following is my private mqtt setting on Pi for your reference:
- in `/etc/mosquitto/mosquitto.conf`
```nginx
pid_file /run/mosquitto/mosquitto.pid

persistence true
persistence_location /var/lib/mosquitto/
log_dest file /var/log/mosquitto/mosquitto.log
password_file /etc/mosquitto/p1.txt
include_dir /etc/mosquitto/conf.d

# this will listen for mqtt on tcp
listener 1883
allow_anonymous false

# this will expect websockets connections
listener 8083
protocol websockets
```
- In `nginx.config`
```nginx
worker_processes  1;

events {
    worker_connections  1024;
}

http {
    include       mime.types;
    default_type  application/octet-stream;
    send_timeout 30;
    sendfile        on;
    keepalive_timeout  65;

    server {
        listen *:443 ssl;
        listen [::]:443 ssl;
        server_name my-private.ddns.net localhost;

        ssl_certificate      /etc/letsencrypt/live/my-private.ddns.net/fullchain.pem;
        ssl_certificate_key  /etc/letsencrypt/live/my-private.ddns.net/privkey.pem;

        location /mqtt {
            proxy_pass http://127.0.0.1:8083;
            proxy_http_version 1.1;
            proxy_set_header X-Real-IP $remote_addr;
            proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
            proxy_set_header Host $host;
            proxy_set_header Upgrade $http_upgrade;
            proxy_set_header Connection "Upgrade";
        }
    }
}
```
If you follow my setting above , the mqtt in the app will be
```ini
URL = my-private.ddns.net
Path = /mqtt
Port = 443
username = xxxx
password = xxxx
```

xthree, 2024-09-28:
Thank you very much for your assistance. I got it working with a ddns pointing to my local Raspberry Pi IP (192.168.x.x), some free SSL certs, and your configs.

If anyone else needs assistance setting any of these things up, feel free to reach out. I'll be happy to help. @samthelion17

samthelion17, 2024-09-28:
> Thank you very much for your assistance. I got it working with a ddns pointing to my local Raspberry Pi IP (192.168.x.x), some free SSL certs, and your configs.
> 
> If anyone else needs assistance setting any of these things up, feel free to reach out. I'll be happy to help. @samthelion17 

I've been following the discussion and I'm glad you got it working. It'd be nice to document the whole process step-by-step so others can benefit. I will set everyting up again soon using @TzuHuanTai's config and will close this issue if (when) everything works as expected.

samthelion17, 2024-09-29:
I managed to get it working with my own domain name.

closed 2024-09-29 by samthelion17

TaciteOFF, 2025-01-29:
I'd love a full tutorial for that because I have the same issue. I saw the comment but I don't have nginx on my VPS so I'm not sure it's relevant