mazupo/pi-webrtc #185: Support: USB Camera

rkenzhebekov, 2024-09-22:
Can I use other cameras connected via USB, or is the raspberry pi based CSI Camera Module required?

Thank you

TzuHuanTai, 2024-09-22:
Yes,you can. My Pi 4B use a usb camera.

TzuHuanTai, 2024-09-22:
I suggest you to run `v4l2-ctl -d /dev/video0 --list-formats` to confirm supported formats of the usb camera in advance. My usb cam only support specific formats, v4l2 can't read it if my settings do not match.

boredom1234, 2024-10-30:
@TzuHuanTai What to use instead of  --use_libcamera to make use for the Web Cam?

TzuHuanTai, 2024-10-30:
Could you please check the support format of your webcam?
And what is your full command and settings right now?

Here is the way I use webcam on pi, fyi

1. Enable lagecy v4l2 driver, [ref](https://github.com/TzuHuanTai/RaspberryPi_WebRTC/wiki/Advanced-Settings#using-the-legacy-v4l2-driver)

2. Check the support format of the webcam and run it.
```console
pi@home-pi-5:~ $ v4l2-ctl -d /dev/video0 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'MJPG' (Motion-JPEG, compressed)
                Size: Discrete 352x288
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 800x600
                        Interval: Discrete 0.050s (20.000 fps)
                Size: Discrete 1280x720
                        Interval: Discrete 0.033s (30.000 fps)
...

pi@home-pi-5:~ $ ./pi_webrtc --device=/dev/video0 --v4l2_format=mjpeg --fps=30 --width=1920 --height=1080 --mqtt_host=xxxx --mqtt_port=8883 --mqtt_username=xxxx --mqtt_password=xxxx --uid=xxxx
Use mjpeg format source in v4l2
[v4l2_utils] Error: fd(26) set ctrl(9963810): Invalid argument
[v4l2_utils] Error: fd(26) set ext ctrl(10029519): Invalid argument

```

Sometime it might show some errors about v4l2 setting while running `pi_webrtc`.
Because not every webcam supplier implement all v4l2 setting support, and `pi_webrtc` always try to set bitrate/roration to the camera initially, but it doesn't bother the streaming.

boredom1234, 2024-11-02:
> Could you please check the support format of your webcam? And what is your full command and settings right now?
> 
> Here is the way I use webcam on pi, fyi
> 
> 1. Enable lagecy v4l2 driver, [ref](https://github.com/TzuHuanTai/RaspberryPi_WebRTC/wiki/Advanced-Settings#using-the-legacy-v4l2-driver)
> 2. Check the support format of the webcam and run it.
> 
> ```
> pi@home-pi-5:~ $ v4l2-ctl -d /dev/video0 --list-formats-ext
> ioctl: VIDIOC_ENUM_FMT
>         Type: Video Capture
> 
>         [0]: 'MJPG' (Motion-JPEG, compressed)
>                 Size: Discrete 352x288
>                         Interval: Discrete 0.033s (30.000 fps)
>                 Size: Discrete 640x480
>                         Interval: Discrete 0.033s (30.000 fps)
>                 Size: Discrete 800x600
>                         Interval: Discrete 0.050s (20.000 fps)
>                 Size: Discrete 1280x720
>                         Interval: Discrete 0.033s (30.000 fps)
> ...
> 
> pi@home-pi-5:~ $ ./pi_webrtc --device=/dev/video0 --v4l2_format=mjpeg --fps=30 --width=1920 --height=1080 --mqtt_host=xxxx --mqtt_port=8883 --mqtt_username=xxxx --mqtt_password=xxxx --uid=xxxx
> Use mjpeg format source in v4l2
> [v4l2_utils] Error: fd(26) set ctrl(9963810): Invalid argument
> [v4l2_utils] Error: fd(26) set ext ctrl(10029519): Invalid argument
> ```
> 
> Sometime it might show some errors about v4l2 setting while running `pi_webrtc`. Because not every webcam supplier implement all v4l2 setting support, and `pi_webrtc` always try to set bitrate/roration to the camera initially, but it doesn't bother the streaming.

Use h264 format source in v4l2
[v4l2_utils] Error: fd(16) set ctrl(9963810): Invalid argument
[v4l2_utils] Error: fd(16) set ext ctrl(10029519): Invalid argument
[v4l2_utils] Error: fd(16) set ext ctrl(10029518): Invalid argument
[v4l2_utils] Error: fd(16) set ext ctrl(10029675): Invalid argument
[v4l2_utils] Error: fd(16) set ext ctrl(10029538): Invalid argument
[v4l2_utils] Error: fd(16) set ext ctrl(10029671): Invalid argument
[v4l2_utils] Error: fd(16) set ext ctrl(10029670): Invalid argument
[v4l2_utils] Error: fd(16) set ext ctrl(10029541): Invalid argument
[v4l2_utils] Error: fd(16) set ext ctrl(10029519): Invalid argument
Segmentation fault

TzuHuanTai, 2025-02-26:
see #237

closed 2025-04-17 by TzuHuanTai