mazupo/pi-webrtc #188: RPI 3B+ Segmentation fault.

LRBin, 2024-09-27:
I'm running pi_webrtc (https://github.com/TzuHuanTai/RaspberryPi_WebRTC/releases/download/v1.0.0/pi_webrtc-1.0.0_pi-os-bookworm.tar.gz) on a Raspberry Pi 3B+ with Debian Bookworm 64-bit system. When I try to preview the video stream on my phone, a segmentation fault occurs. The relevant debug log is as follows:

root@raspberrypi:/home/test# gdb ./pi_webrtc
GNU gdb (Debian 13.1-3) 13.1
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "aarch64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./pi_webrtc...
(No debugging symbols found in ./pi_webrtc)
(gdb) run --device=/dev/video0 --fps=20 --width=640 --height=360 --v4l2_format=h264 --hw_accel --mqtt_host=xxxxxxxxxxxx.hivemq.cloud --mqtt_port=8883 --mqtt_username=xxxxx --mqtt_password=xxxxx --uid=xxxxxxx
Starting program: /home/xxxx/pi_webrtc --device=/dev/video0 --fps=20 --width=640 --height=360 --v4l2_format=h264 --hw_accel --mqtt_host=xxxxxxxxxxxx.hivemq.cloud --mqtt_port=8883 --mqtt_username=xxxxxx --mqtt_password=xxxxxxxxx --uid=xxxxxxxxxxx
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
Use h264 format source in v4l2
[New Thread 0x7fed34e4c0 (LWP 2001)]
[New Thread 0x7fe4b3e4c0 (LWP 2002)]
[New Thread 0x7fecb3e4c0 (LWP 2003)]
[New Thread 0x7fec32e4c0 (LWP 2004)]
[New Thread 0x7fec21e4c0 (LWP 2005)]
[New Thread 0x7fe7ffe4c0 (LWP 2006)]
[New Thread 0x7fec10e4c0 (LWP 2007)]
[New Thread 0x7fe77ee4c0 (LWP 2008)]
[New Thread 0x7fe76de4c0 (LWP 2009)]
[New Thread 0x7fe6ece4c0 (LWP 2010)]
[New Thread 0x7fe6b284c0 (LWP 2011)]
[New Thread 0x7fe6a184c0 (LWP 2012)]
[New Thread 0x7fe685e4c0 (LWP 2013)]
[New Thread 0x7fe670e4c0 (LWP 2015)]
[New Thread 0x7fe5efe4c0 (LWP 2016)]
[New Thread 0x7fe5dee4c0 (LWP 2025)]
=> Set sdp success!
=> Set sdp success!
[New Thread 0x7fe55de4c0 (LWP 2026)]
[New Thread 0x7fe54ce4c0 (LWP 2027)]
[New Thread 0x7fe53be4c0 (LWP 2028)]
[New Thread 0x7fe52ae4c0 (LWP 2029)]
[Thread 0x7fe55de4c0 (LWP 2026) exited]
[Thread 0x7fe54ce4c0 (LWP 2027) exited]
[Thread 0x7fe53be4c0 (LWP 2028) exited]
[Thread 0x7fe52ae4c0 (LWP 2029) exited]
[Thread 0x7fe670e4c0 (LWP 2015) exited]

Thread 4 "Thread 0x0x1329" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fecb3e4c0 (LWP 2003)]
0x0000000000e80d20 in V4L2Capture::GetI420Frame() ()
(gdb) bt full
#0  0x0000000000e80d20 in V4L2Capture::GetI420Frame() ()
No symbol table info available.
#1  0x0000000000418470 in std::_Function_handler<void (std::shared_ptr<DataChannelSubject>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >), Conductor::CreatePeerConnection()::$_5>::_M_invoke(std::_Any_data const&, std::shared_ptr<DataChannelSubject>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&) ()
No symbol table info available.
#2  0x000000000043fdf4 in std::_Function_handler<void (std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >), RtcPeer::SubscribeCommandChannel(CommandType, std::function<void (std::shared_ptr<DataChannelSubject>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)>)::$_6>::_M_invoke(std::_Any_data const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&) ()
No symbol table info available.
#3  0x0000000000430ab8 in DataChannelSubject::Next(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) ()
No symbol table info available.
#4  0x0000000000430670 in DataChannelSubject::OnMessage(webrtc::DataBuffer const&) ()
No symbol table info available.
#5  0x00000000004556a4 in rtc::Thread::Dispatch(absl::AnyInvocable<void () &&>) ()
No symbol table info available.
#6  0x00000000004540d4 in rtc::Thread::ProcessMessages(int) ()
No symbol table info available.
#7  0x0000000000455cc8 in rtc::Thread::PreRun(void*) ()
No symbol table info available.
#8  0x0000007ff624ee30 in start_thread (arg=0x7fffffe887) at ./nptl/pthread_create.c:442
        ret = <optimized out>
        pd = 0x7fffffe887
        out = <optimized out>
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {549432059072, 549755808048, 549755807878, 0, 8445312, 549755807879, 549423611904, 549432059072, 549423611904, 549432060928, 549432056928, 
                1605013229623844164, 0, 1605013229449194936, 0, 0, 0, 0, 0, 0, 0, 0}, mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x0, 0x0}, data = {prev = 0x0, cleanup = 0x0, 
              canceltype = 0}}}
        not_first_call = 0
#9  0x0000007ff62b7adc in thread_start () at ../sysdeps/unix/sysv/linux/aarch64/clone.S:79
No locals.
(gdb)

TzuHuanTai, 2024-09-27:
I tried it on debug mode. It seems to RPI driver's bug again as I mentioned in the readme. :( 

`
Hint 2: I noticed that when I set 1920x1080, the hardware decoder firmware changes it to 1920x1088, but the isp/encoder does not adjust in the 6.6.31 kernel. This leads to memory being out of range. However, if I set 1920x1088, all works fine.
`

Even if 640x360 is set to the driver, only the hw decoder will reallocate to 640x368 automatically, but the hw scaler and encoder are without this issue. I don't know why the hw decoder's driver will round to multiples of 16. You can set `--width=640 --height=368`, it'll be fine.
If the height must be `360`, please remove the `--hw_accel` to run the video pipeline in software mode.

```bash
$ /home/pi/IoT/RaspberryPi_WebRTC/build/pi_webrtc --device=/dev/video0 --fps=30 \
--width=640 --height=360 --v4l2_format=mjpeg --hw_accel \
--mqtt_host=xxxxxx --mqtt_port=8883 --mqtt_username=xxxx--mqtt_password=xxxx--uid=home-pi-4b
Use mjpeg format source in v4l2
[conductor] network thread start: success!
[conductor] worker thread start: success!
[conductor] signaling thread start: success!
[v4l2_utils] Open file /dev/video0 fd(26) success!
[v4l2_utils] fd(26) driver 'bcm2835 mmal' on card 'mmal service 16.1' in splane mode
[v4l2_capture]   Fps: 30
[v4l2_capture]   Rotation: 0
[v4l2_utils] fd(26) original formats: MJPG(640x360)
[v4l2_utils] fd(26) latest format: MJPG(640x360)
[v4l2_utils] fd(26) query buffer at 0x7fa808a000 (length: 235520)
[v4l2_utils] fd(26) query buffer at 0x7fa8050000 (length: 235520)
[v4l2_utils] fd(26) query buffer at 0x7fa8016000 (length: 235520)
[v4l2_utils] fd(26) query buffer at 0x7fa26c2000 (length: 235520)
[v4l2_utils] Open file /dev/video10 fd(27) success!
[v4l2_utils] fd(27) driver 'bcm2835-codec' on card 'bcm2835-codec-decode' in mplane mode
[v4l2_utils] fd(27) original formats: H264(32x32)
[v4l2_utils] fd(27) latest format: MJPG(640x360)        // <===============
[v4l2_utils] fd(27) query buffer at 0x7fa2642000 (length: 524288)
[v4l2_utils] fd(27) query buffer at 0x7fa25c2000 (length: 524288)
[v4l2_utils] fd(27) driver 'bcm2835-codec' on card 'bcm2835-codec-decode' in mplane mode
[v4l2_utils] fd(27) original formats: YU12(640x368)    // <===============
[v4l2_utils] fd(27) latest format: YU12(640x368)         // <===============
[v4l2_utils] fd(27) export dma at fd(28)
[v4l2_utils] fd(27) query buffer at 0x7fa256b000 (length: 353280)
[v4l2_utils] fd(27) export dma at fd(29)
[v4l2_utils] fd(27) query buffer at 0x7fa2514000 (length: 353280)
[v4l2_utils] Open file /dev/video12 fd(30) success!
[v4l2_utils] fd(30) driver 'bcm2835-codec' on card 'bcm2835-codec-isp' in mplane mode
[v4l2_utils] fd(30) original formats: BA81(32x32)
[v4l2_utils] fd(30) latest format: YU12(640x360)
[v4l2_utils] fd(30) driver 'bcm2835-codec' on card 'bcm2835-codec-isp' in mplane mode
[v4l2_utils] fd(30) original formats: YUYV(32x32)
[v4l2_utils] fd(30) latest format: YU12(640x360)
[v4l2_utils] fd(30) export dma at fd(31)
[v4l2_utils] fd(30) query buffer at 0x7fa22bd000 (length: 345600)
[v4l2_utils] fd(30) export dma at fd(32)
[v4l2_utils] fd(30) query buffer at 0x7fa2268000 (length: 345600)
[main] Recorder is not started!
[rtc_peer] The data channel is established successfully.
[conductor] Peer connection(1) is created!
[main] Wait for signaling!
[mqtt_service] Successfully subscribed to topic: home-pi-4b/sdp/+/offer
[mqtt_service] Successfully subscribed to topic: home-pi-4b/ice/+/offer
[mqtt_service] MQTT service is ready.

```

LRBin, 2024-09-29:
I tried setting it to 1920x1088 and setting gpu_mem to 256. Thanks.

closed 2024-09-29 by TzuHuanTai