mazupo/pi-webrtc #204: Support: Multiple Cameras

boredom1234, 2024-11-03:
Is it possible to connect more than one camera to a single Raspberry Pi 4 that has been piped for multiple streaming instances?

TzuHuanTai, 2024-11-07:
Connecting multiple cameras to the Raspberry Pi definitely works. For example, two cameras will be detected and listed as /dev/video0 and /dev/video1. However, I was only able to get this working using the legacy v4l2 driver. I haven't researched how to do this in libcamera mode yet. I think you could try running two separate `pi_webrtc` processes in [v4l2 mode](https://github.com/TzuHuanTai/RaspberryPi-WebRTC/wiki#v4l2).
```
./pi_webrtc --device=/dev/video0 ...
./pi_webrtc --device=/dev/video1 ...
```

TzuHuanTai, 2025-02-26:
see #237

closed 2025-04-17 by TzuHuanTai