mazupo/pi-webrtc #215: stream keeps dying periodically

davidus-sk, 2024-12-15:
Stream works for a few minutes then dies and this is what the binary outputs:

=> Set sdp success!
=> Set sdp success!
=> Set sdp success!
=> Set sdp success!
=> Set sdp success!
=> Set sdp success!
[v4l2_utils] Error: fd(61) queue buffer(10): Bad address

[v4l2_codec] Error: QueueBuffer V4L2_BUF_TYPE_VIDEO_OUTPUT_MPLANE. fd(61) at index 0
[v4l2_utils] Error: fd(61) queue buffer(10): Bad address

[v4l2_codec] Error: QueueBuffer V4L2_BUF_TYPE_VIDEO_OUTPUT_MPLANE. fd(61) at index 1
[v4l2_utils] Error: fd(62) queue buffer(10): Bad address

[v4l2_codec] Error: QueueBuffer V4L2_BUF_TYPE_VIDEO_OUTPUT_MPLANE. fd(62) at index 0
[v4l2_utils] Error: fd(62) queue buffer(10): Bad address

[v4l2_codec] Error: QueueBuffer V4L2_BUF_TYPE_VIDEO_OUTPUT_MPLANE. fd(62) at index 1
[v4l2_utils] Error: fd(68) turn on stream: No such file or directory
[v4l2_utils] Error: fd(60) queue buffer(10): Bad address

[v4l2_codec] Error: QueueBuffer V4L2_BUF_TYPE_VIDEO_OUTPUT_MPLANE. fd(60) at index 0
[v4l2_utils] Error: fd(68) queue buffer(10): Bad address

[v4l2_codec] Error: QueueBuffer V4L2_BUF_TYPE_VIDEO_OUTPUT_MPLANE. fd(68) at index 0
[v4l2_utils] Error: fd(60) queue buffer(10): Bad address

[v4l2_codec] Error: QueueBuffer V4L2_BUF_TYPE_VIDEO_OUTPUT_MPLANE. fd(60) at index 1
[v4l2_utils] Error: fd(68) queue buffer(10): Bad address

[v4l2_codec] Error: QueueBuffer V4L2_BUF_TYPE_VIDEO_OUTPUT_MPLANE. fd(68) at index 1
=> Set sdp success!
=> Set sdp success!
[v4l2_utils] Error: fd(145) turn on stream: No such file or directory
=> Set sdp success!
=> Set sdp success!
[v4l2_utils] Error: fd(158) turn on stream: No such file or directory
=> Set sdp success!
=> Set sdp success!
[v4l2_utils] Error: fd(137) turn on stream: No such file or directory
=> Set sdp success!
=> Set sdp success!
[v4l2_utils] Error: fd(137) turn on stream: No such file or directory
=> Set sdp success!
=> Set sdp success!
=> Set sdp success!
=> Set sdp success!
[v4l2_utils] Error: fd(160) turn on stream: No such file or directory
[v4l2_utils] Error: fd(69) turn on stream: No such file or directory
[v4l2_utils] Error: fd(60) turn on stream: No such file or directory
[v4l2_utils] Error: fd(69) turn on stream: No such file or directory
[v4l2_utils] Error: fd(70) turn on stream: No such file or directory
[v4l2_utils] Error: fd(62) turn on stream: Timer expired
[v4l2_utils] Error: fd(70) set ext ctrl(10029541): Invalid argument
=> Set sdp success!
=> Set sdp success!
[v4l2_utils] Error: fd(62) turn on stream: Timer expired

Restarting the binary fixes things temporarily until it happens again.

TzuHuanTai, 2024-12-16:
What is your full command for `pi_webrtc`? Are you using the `--libcamera` flag?

I see `Raspberry Pi Compute Module 3` and `kernel 6.6.51` in your dmsg. Raspberry Pi officials had mentioned that Broadcom camera stack will not support latest camera module, include module 3. [ref](https://forums.raspberrypi.com/viewtopic.php?t=362707)

davidus-sk, 2024-12-16:
Thanks for your response:

Full command: `pi_webrtc --use_libcamera --fps=10 --width=1280 --height=720 --hw_accel --no_audio --mqtt_host=<server> --mqtt_port=8883 --mqtt_username=<user> --mqtt_password=<pass> --uid=<uuid>`

I am using an Arducam camera, not any of the official pi cam module(s).

I ran `rpi-update` on the CM3 which I think fixed something, but now I am getting bunch of these:

[ 2864.924229] bcm2835-codec bcm2835-codec: bcm2835_codec_start_streaming: Failed enabling i/p port, ret -2
[ 2893.945228] bcm2835-codec bcm2835-codec: bcm2835_codec_start_streaming: Failed enabling i/p port, ret -2
[ 2941.055586] bcm2835-codec bcm2835-codec: bcm2835_codec_start_streaming: Failed enabling i/p port, ret -2
[ 2941.185915] bcm2835-codec bcm2835-codec: bcm2835_codec_start_streaming: Failed enabling i/p port, ret -2
[ 2962.704167] bcm2835-codec bcm2835-codec: bcm2835_codec_start_streaming: Failed enabling i/p port, ret -2
[ 2992.734680] bcm2835-codec bcm2835-codec: bcm2835_codec_start_streaming: Failed enabling i/p port, ret -2
[ 3019.842865] bcm2835-codec bcm2835-codec: bcm2835_codec_start_streaming: Failed enabling i/p port, ret -2
[ 3041.367150] bcm2835-codec bcm2835-codec: bcm2835_codec_start_streaming: Failed enabling i/p port, ret -2
[ 3097.282571] bcm2835-codec bcm2835-codec: bcm2835_codec_start_streaming: Failed enabling i/p port, ret -2

davidus-sk, 2024-12-16:
After about 2 hours it the binary outputs this and streaming no longer works:

[v4l2_utils] Error: fd(57) dequeue buffer: Invalid argument
[v4l2_utils] Error: fd(65) dequeue buffer: Invalid argument
[v4l2_utils] Error: fd(57) dequeue buffer: Invalid argument
[v4l2_utils] Error: fd(65) dequeue buffer: Invalid argument
[v4l2_utils] Error: fd(57) dequeue buffer: Invalid argument
[v4l2_utils] Error: fd(65) dequeue buffer: Invalid argument
[v4l2_utils] Error: fd(57) dequeue buffer: Invalid argument
[v4l2_utils] Error: fd(65) dequeue buffer: Invalid argument
[v4l2_utils] Error: fd(57) dequeue buffer: Invalid argument
[v4l2_utils] Error: fd(65) dequeue buffer: Invalid argument
[v4l2_utils] Error: fd(57) dequeue buffer: Invalid argument
[v4l2_utils] Error: fd(65) dequeue buffer: Invalid argument
[v4l2_utils] Error: fd(57) dequeue buffer: Invalid argument
[v4l2_utils] Error: fd(65) dequeue buffer: Invalid argument
[v4l2_utils] Error: fd(57) dequeue buffer: Invalid argument
[v4l2_utils] Error: fd(65) dequeue buffer: Invalid argument
[v4l2_utils] Error: fd(57) dequeue buffer: Invalid argument
[v4l2_utils] Error: fd(65) dequeue buffer: Invalid argument
[v4l2_utils] Error: fd(57) dequeue buffer: Invalid argument
[v4l2_utils] Error: fd(65) dequeue buffer: Invalid argument
[v4l2_utils] Error: fd(57) dequeue buffer: Invalid argument
[v4l2_utils] Error: fd(65) dequeue buffer: Invalid argument
[v4l2_utils] Error: fd(57) dequeue buffer: Invalid argument
[v4l2_utils] Error: fd(65) dequeue buffer: Invalid argument
[v4l2_utils] Error: fd(57) dequeue buffer: Invalid argument
[v4l2_utils] Error: fd(65) dequeue buffer: Invalid argument
[v4l2_utils] Error: fd(57) dequeue buffer: Invalid argument
[v4l2_utils] Error: fd(65) dequeue buffer: Invalid argument
[v4l2_utils] Error: fd(57) dequeue buffer: Invalid argument
[v4l2_utils] Error: fd(65) dequeue buffer: Invalid argument

TzuHuanTai, 2024-12-16:
Could you please remove `--hw_accel` and take a try under software encoding mode?

The error log from `[v4l2_utils]` is kind of weird. There're only two function use `v4l2_utils`, one is reading camera via legacy v4l2 stack, another is hardware encoder. Your command already apply `--use_libcamera`, so the problem must come from hw encoder.

davidus-sk, 2024-12-16:
Thanks for the suggestion. I took `--hw_accel` out but now it won't do anything at all. It wont stream or even connect to MQTT.

The executable hangs with low CPU utilization (at like 5%) and the log file ends here:

[1:26:00.118424808] [26907]  INFO Camera camera_manager.cpp:325 libcamera v0.3.2+99-1230f78d
[1:26:00.284158309] [26915]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[1:26:00.289021286] [26915]  INFO RPI vc4.cpp:447 Registered camera /base/soc/i2c0mux/i2c@0/imx708@1a to Unicam device /dev/media1 and ISP device /dev/media0
[libcamera_capturer] camera id: /base/soc/i2c0mux/i2c@0/imx708@1a
[libcamera_capturer]   Fps: 10
[libcamera_capturer]   AutoFocus: 2
[libcamera_capturer]   Rotation: 0
[libcamera_capturer] camera original format: 1920x1080-YUV420
[libcamera_capturer] camera validated format: 1280x720-YUV420.
[libcamera_capturer]   width: 1280, height: 720, stride: 1280
[1:26:00.290127839] [26907]  INFO Camera camera.cpp:1197 configuring streams: (0) 1280x720-YUV420
[1:26:00.290717625] [26915]  INFO RPI vc4.cpp:622 Sensor: /base/soc/i2c0mux/i2c@0/imx708@1a - Selected sensor format: 1536x864-SBGGR10_1X10 - Selected unicam format: 1536x864-pBAA

^  there are no more lines in the log even if I let it sit for an hour. When I `kill` the process and restart it, it does the same thing.

davidus-sk, 2024-12-19:
> Could you please remove `--hw_accel` and take a try under software encoding mode?
>
> The error log from `[v4l2_utils]` is kind of weird. There're only two function use `v4l2_utils`, one is reading [...]

I have tried it without `--hw_accel` on the CM3 and it is _way_ more stable. Still you get error messages, but at some point I got itt to stream for 7 hours non-stop.

TzuHuanTai, 2024-12-20:
I've never tried to watch a live stream for more than 3 hours without a break.
I don't have a CM3(BCM2837) so I'm testing on my pi zero 2w (BCM2710A1) and `--hw_accel`.
I'll let the client play the stream for 24hr and see what happens.
The cpu usage is steady between 15 and 20% with the help of hardware acceleration, hopefully it won't overheat.

```console
pi@home-pi-zero2w:~ $ uname -a
Linux home-pi-zero2w 6.6.51+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux
pi@home-pi-zero2w:~ $ /home/pi/pi_webrtc --fps=30 --width=1280 --height=720 --mqtt_host=xxxx --mqtt_port=8883 --mqtt_username=xxxx --mqtt_password=xxxx --uid=home-pi-zero2w --use_libcamera --no_audio --hw_accel
```

[image]

TzuHuanTai, 2024-12-21:
My client-side application has been streaming from the Pi Zero 2W for over 24 hours uninterrupted without any issues.
I believe it can continue working for another 24 hours. I suspect your issue might be related to the hardware or driver level.

closed 2025-04-17 by TzuHuanTai