mazupo/pi-webrtc #216: [mqtt_service] Error: Lookup error.

davidus-sk, 2024-12-16:
When the Pi boots up and `pi_webrtc` is launched before the network is ready, (obviously) the binary cannot reach the provided MQTT server thus it fails. The binary writes this into the error log:

[mqtt_service] Error: Lookup error.

The problem is that there is no retry mechanism. The binary is running but not doing anything even when the network becomes available. A retry mechanism would be nice. When I `kill` `pi_webrtc` and bring it back up, it start to work correctly. However without a manual intervention it just sits there doing nothing.

TzuHuanTai, 2024-12-16:
I know reconnection is a nessary feature, servers or internet might shut down as well. So it's not only for booting up.
You can use `After=network-online.target` ([ref](https://github.com/TzuHuanTai/RaspberryPi-WebRTC/wiki/Advanced-Settings#2-in-order-to-run-pi_webrtc-and-ensure-it-starts-automatically-on-reboot)) while rebooting before I implement the feature.

davidus-sk, 2024-12-16:
Thank you. I will use your suggestion or implement my own to wait till network is available.

TzuHuanTai, 2024-12-19:
It's implemented in `v1.0.3`.

closed 2024-12-19 by TzuHuanTai