mazupo/pi-webrtc #220: same mqtt client id - Unable to reconnect

meriksk, 2024-12-19:
v1.0.3
for my web-app I need to store mqtt client id in the cache for a short time (i.e. 1 minute). Stream works well for the first time. Then I am unable to connect with a new RTCPeerConnection but the same mqtt client id.  

**command:**
`pi_webrtc --use_libcamera --fps=10 --width=1280 --height=960 --hw_accel --no_audio --mqtt_host=**** --mqtt_port=8883 --mqtt_username=**** --mqtt_password=**** --uid=device-100000003a0a2f6e`

console output:

**first try:**

```
[0:07:06.813328372] [1576]  INFO Camera camera_manager.cpp:325 libcamera v0.3.2+99-1230f78d
[0:07:06.848779699] [1584]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:07:06.850852454] [1584]  INFO RPI vc4.cpp:447 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media0 and ISP device /dev/media2
[libcamera_capturer] camera id: /base/soc/i2c0mux/i2c@1/imx708@1a
[libcamera_capturer] camera validated format: 1280x960-YUV420.
[libcamera_capturer]   width: 1280, height: 960, stride: 1280
[0:07:06.851493039] [1576]  INFO Camera camera.cpp:1197 configuring streams: (0) 1280x960-YUV420
[0:07:06.851839554] [1584]  INFO RPI vc4.cpp:622 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 1536x864-SBGGR10_1X10 - Selected unicam format: 1536x864-pBAA
=> Set sdp success!
=> Set sdp success!

... stream is connected ...

```
**reloading the page - initializing new RTCPeerConnection (same mqtt client id):** 

```
... not response to offer ...
... no console output for 30s ...

=> Set sdp failed! Failed to set remote offer sdp: Called in wrong state: closed
[rtc_peer] Error: INVALID_STATE; PeerConnection cannot create an answer in a state other than have-remote-offer or have-local-pranswer.
[rtc_peer] Error: Failed to apply the received candidate!
[rtc_peer] Error: Failed to apply the received candidate!
[rtc_peer] Error: Failed to apply the received candidate!

... no console output for 30s, still unable to connect ...

=> Set sdp success!
=> Set sdp success!

```

after approx. 60s of the first connection I am able to connect again

TzuHuanTai, 2024-12-20:
Oh, because I use a hash map to cache a pair of an MQTT client and an RTC peer in `v1.0.3`, and the hash map is refreshed very 60s. 
I assumed mqtt client would disconnect after webrtc connection was established, but I didn't consider the possibility of having the same client ID.

TzuHuanTai, 2024-12-22:
The updated code has been integrated into the main branch, and this issue will be resolved in the next release.

TzuHuanTai, 2024-12-29:
It's fixed in `v1.0.4`.

PR #221 (2024-12-21): fix: reconnect while using the same mqtt id
fix #220
closed 2024-12-29 by TzuHuanTai