mazupo/pi-webrtc #236: picamera3 error on latest updated Pi OS Bookworm 64bit

GregLeichner, 2025-02-07:
Latest Pi OS Bookworm 64 bit release via the Raspberry Pi Imager on a Pi 3B and after standard post-OS install steps:

sudo apt update
sudo apt upgrade

Running pi_webrtc with:

./pi_webrtc --fps=30 --width=1280 --height=720 --http_port=8080 --uid=home-pi-3 --use_libcamera

Results in:

./pi_webrtc: symbol lookup error: /lib/aarch64-linux-gnu/libcamera.so.0.3: undefined symbol: _ZN7libpisp22compute_optimal_strideER24pisp_image_format_config

I attempted to manually uninstall and re-install libcamera0.3, but the error remains the same.

TzuHuanTai, 2025-02-08:
Can you run the following commands on your Pi and share the outputs?
```
uname -a
ldd ./pi_webrtc | grep libcamera
sudo ldconfig -v | grep libcamera
```

Here are my outputs:
```console
pi@home-pi-5:~ $ uname -a
Linux home-pi-5 6.6.51+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux
pi@home-pi-5:~ $ sudo ldconfig -v | grep libcamera
        libcamera-base.so.0.3 -> libcamera-base.so.0.3.2
        rpicam_app.so.1.5.2 -> libcamera_app.so
        libcamera.so.0.3 -> libcamera.so.0.3.2
pi@home-pi-5:~ $ ldd /home/pi/IoT/RaspberryPi_WebRTC/build/pi_webrtc | grep libcamera
        libcamera.so.0.3 => /lib/aarch64-linux-gnu/libcamera.so.0.3 (0x00007fff37ac0000)
        libcamera-base.so.0.3 => /lib/aarch64-linux-gnu/libcamera-base.so.0.3 (0x00007fff37a70000)
```

The latest `pi_webrtc` was compiled with [`libcamera0.3.2`](https://github.com/TzuHuanTai/RaspberryPi-WebRTC/actions/runs/12984552012/job/36207883653#step:5:35). If the needed lib doesn't match, please update to latest `pi_webrtc` .

GregLeichner, 2025-02-08:
I verified that it does run correctly if you do not do the apt upgrade after the initial OS install.

GregLeichner, 2025-02-08:
After re-applying the apt upgrade:

greg@raspberrypi:~ $ uname -a
Linux raspberrypi 6.6.51+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux

greg@raspberrypi:~ $ ldd ./pi_webrtc | grep libcamera
        libcamera.so.0.3 => /lib/aarch64-linux-gnu/libcamera.so.0.3 (0x0000007fa3b00000)
        libcamera-base.so.0.3 => /lib/aarch64-linux-gnu/libcamera-base.so.0.3 (0x0000007fa3ab0000)

greg@raspberrypi:~ $ sudo ldconfig -v | grep libcamera
ldconfig: Can't stat /usr/local/lib/aarch64-linux-gnu: No such file or directory
ldconfig: Path `/usr/lib/aarch64-linux-gnu' given more than once
(from /etc/ld.so.conf.d/aarch64-linux-gnu.conf:4 and /etc/ld.so.conf.d/aarch64-linux-gnu.conf:3)
ldconfig: Path `/lib/aarch64-linux-gnu' given more than once
(from <builtin>:0 and /etc/ld.so.conf.d/aarch64-linux-gnu.conf:3)
ldconfig: Path `/usr/lib/aarch64-linux-gnu' given more than once
(from <builtin>:0 and /etc/ld.so.conf.d/aarch64-linux-gnu.conf:3)
ldconfig: Path `/usr/lib' given more than once
(from <builtin>:0 and <builtin>:0)
        libcamera-base.so.0.4 -> libcamera-base.so.0.4.0
ldconfig: /lib/aarch64-linux-gnu/ld-linux-aarch64.so.1 is the dynamic linker, ignoring

        rpicam_app.so.1.6.0 -> libcamera_app.so
        libcamera-base.so.0.3 -> libcamera-base.so.0.3.2
        libcamera.so.0.3 -> libcamera.so.0.3.2
        libcamera.so.0.4 -> libcamera.so.0.4.0

TzuHuanTai, 2025-02-08:
I think I know where the issue comes from.
For `libcamera-0.3`:
The initial version was `0.3.2+rpt20240927-1`.
The reinstalled version is `0.3.2+rpt20241119-1`.
The API seems to have changed, even though both are labeled as `0.3.2`. :(

GregLeichner, 2025-02-08:
Thanks for following up on this so quick!

davidus-sk, 2025-02-10:
I am on Pi Zero 2W `6.6.51+rpt-rpi-v8`

libcamera-hello works:

`root@push-cam:/app/bodycam2/camera/stream# libcamera-hello
[0:26:55.537610607] [1439]  INFO Camera camera_manager.cpp:327 libcamera v0.4.0+50-83cb8101`

But pi_webrtc does not:

`./pi_webrtc_1.0.5: symbol lookup error: /lib/aarch64-linux-gnu/libcamera.so.0.3: undefined symbol: _ZN7libpisp22compute_optimal_strideER24pisp_image_format_config`

TzuHuanTai, 2025-02-10:
@davidus-sk `libcamera-hello` does not work if `libcamera0.3` is updated only.

Solving this problem is quite a dilemma.

The latest released Raspberry Pi OS is on November 19th 2024. `pi-webrtc` link to its default libcamera version, `0.3.2+rpt20240927-1`. Ideally, if the symbol has been changed, **it no longer be labeled as `0.3.2`**. 

Even after executing `sudo apt full-upgrade`, both `0.4.0` and `0.3.2` versions still exist on the Pi. The root cause is the latest  `0.3.2+rpt20241119-1` doesn't support backwards compatibility to `0.3.2+rpt20240927-1`.
No matter link which `0.3.2` version, there will always be someone who can't use it.

The best solution I can think of right now is to avoid updating for now and wait until the latest stable OS version is released, then re-release `pi_webrtc` linked to the `libcamera0.4`. Do you have any suggestions?

davidus-sk, 2025-02-10:
> libcamera-hello does not work if libcamera0.3 is updated only.

yes this is correct, it does however work with `0.4.0` when you force upgrading the libcamera package.

I had to roll back to the September 2024 version of libcamera `0.3.2`.

I think your solution is good.

GregLeichner, 2025-02-10:
I would also agree with davidus-sk

TzuHuanTai, 2025-02-23:
I released the `v1.0.6-rc.0`, it use all the latest packages. 
The stable `v1.0.6` will release only when pi os update to the next stable version.

closed 2025-04-17 by TzuHuanTai