mazupo/pi-webrtc #237: Multiple cameras - How to select the one you want?

davidus-sk, 2025-02-11:
Pi5 has two CSI ports and Pi4 or even Pi0 2W is capable of interacting with additional cameras via USB. I think it would be useful to be to select which camera we want to stream (and possibly through multiple instances of `pi_webrtc` send multiple streams from one Pi).

Thoughts?

TzuHuanTai, 2025-02-11:
Such as #204.
Disable auto-detecting camera on the pi by using V4L2 driver, then the `--device` flag to select which camera you want.
This way also works for `Bookworm`.

davidus-sk, 2025-02-13:
Thanks for the feedback @TzuHuanTai. I am glad it exists.

One more idea from me: Have you thought about adding a support for an OSD? Let's say I want to insert temperature or battery level or any other useful information into the video stream and have it show up as an overlay. I think this would be useful for home monitoring, etc.

TzuHuanTai, 2025-02-13:
I did think about it before.
But this Pi WebRTC can be used in various scenarios, and not everyone needs OSD. So, I think it's better to keep the streaming function purely focused on streaming. If OSD is needed, it shouldn't be difficult to call an API to retrieve data from the Raspberry Pi and then draw it on a canvas on top of the video element.

davidus-sk, 2025-02-14:
Understood and thank you. Closing this.

closed 2025-02-14 by davidus-sk

TzuHuanTai, 2025-02-23:
I found an issue that the camera module v3 must use the libcamera stack on pi. I'll support selecting specific camera when using libcamera.

davidus-sk, 2025-02-24:
What about non-CSI cameras, like a USB camera or some other video device accessible via /dev/videoX, can these be selected?

TzuHuanTai, 2025-02-24:
That works already. 
Only Two CSI camera doesn't work for now.

TzuHuanTai, 2025-02-26:
It's released in `v1.0.6-rc.1`, please use the latest flag `--camera`.

`--camera=libcamera:0` for the csi camera, and remove the depricated `--use_libcamera` flag.
`--camera=v4l2:0` for the usb camera at `/dev/video0`.

davidus-sk, 2025-02-26:
You're a magician! Thank you.