mazupo/pi-webrtc #251: Error parsing arguments: unrecognised option '--http_port=8080'

urbinek, 2025-03-01:
Pretty much as in title.

I've installed `pi_webrtc` from `pi_webrtc-v1.0.6-rc.1_MQTT_raspios-bookworm-arm64.tar.gz` following instructions in main readme, and try to run my my USB camera with 
```Bash`
urbinek@zero:~ $ ./pi_webrtc \
  --camera=v4l2:0  \
  --v4l2_format=h264  \
  --fps=30  \
  --width=1280  \
   --height=720 \
  --hw_accel  \
  --no_audio  \
  --http_port=8080  \
  --uid=zero
```
Returning with exit code `1`:
```
Error parsing arguments: unrecognised option '--http_port=8080'
```

TzuHuanTai, 2025-03-01:
You downloaded `MQTT` version, try the package named `HTTP` in release page.

urbinek, 2025-03-01:
Incredible how technology works, when used right :) 

http versions is starting, but I can't see cammera feed
`urbinek@zero:~ $ ./pi_webrtc   --camera=v4l2:0    --v4l2_format=mjpeg    --fps=30    --width=1280     --height=720      --no_audio    --http_port=8080    --uid=zero`
```
Using V4L2: mjpeg
Using V4L2, ID: 0
[v4l2_utils] Error: fd(16) set ctrl(9963810): Invalid argument
[v4l2_utils] Error: fd(16) set ext ctrl(10029519): Invalid argument
[http_service] Http server is running on http://*:8080
```

Upon entering http://zero:8080 I have only
```
Without content type.
```

`$ v4l2-ctl -d /dev/v4l/by-id/usb-046d_Logitech_BRIO_504B397D-video-index0  --all`
```
Driver Info:
        Driver name      : uvcvideo
        Card type        : Logitech BRIO
        Bus info         : usb-3f980000.usb-1
        Driver version   : 6.6.74
        Capabilities     : 0x84a00001
                Video Capture
                Metadata Capture
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps      : 0x04200001
                Video Capture
                Streaming
                Extended Pix Format
Media Driver Info:
        Driver name      : uvcvideo
        Model            : Logitech BRIO
        Serial           : 504B397D
        Bus info         : usb-3f980000.usb-1
        Media version    : 6.6.74
        Hardware revision: 0x00000017 (23)
        Driver version   : 6.6.74
Interface Info:
        ID               : 0x03000002
        Type             : V4L Video
Entity Info:
        ID               : 0x00000001 (1)
        Name             : Logitech BRIO
        Function         : V4L2 I/O
        Flags            : default
        Pad 0x0100000d   : 0: Sink
          Link 0x02000029: from remote pad 0x1000010 of entity 'Processing 3' (Video Pixel Formatter): Data, Enabled, Immutable
Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
        Width/Height      : 1280/720
        Pixel Format      : 'MJPG' (Motion-JPEG)
        Field             : None
        Bytes per Line    : 0
        Size Image        : 1843200
        Colorspace        : sRGB
        Transfer Function : Rec. 709
        YCbCr/HSV Encoding: ITU-R 601
        Quantization      : Default (maps to Full Range)
        Flags             :
Crop Capability Video Capture:
        Bounds      : Left 0, Top 0, Width 1280, Height 720
        Default     : Left 0, Top 0, Width 1280, Height 720
        Pixel Aspect: 1/1
Selection Video Capture: crop_default, Left 0, Top 0, Width 1280, Height 720, Flags:
Selection Video Capture: crop_bounds, Left 0, Top 0, Width 1280, Height 720, Flags:
Streaming Parameters Video Capture:
        Capabilities     : timeperframe
        Frames per second: 30.000 (30/1)
        Read buffers     : 0

User Controls

                     brightness 0x00980900 (int)    : min=0 max=255 step=1 default=-8193 value=128
                       contrast 0x00980901 (int)    : min=0 max=255 step=1 default=57343 value=128
                     saturation 0x00980902 (int)    : min=0 max=255 step=1 default=57343 value=128
        white_balance_automatic 0x0098090c (bool)   : default=1 value=1
                           gain 0x00980913 (int)    : min=0 max=255 step=1 default=57343 value=0
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=2 value=1 (50 Hz)
                                0: Disabled
                                1: 50 Hz
                                2: 60 Hz
      white_balance_temperature 0x0098091a (int)    : min=2000 max=7500 step=10 default=61432 value=5680 flags=inactive
                      sharpness 0x0098091b (int)    : min=0 max=255 step=1 default=57343 value=128
         backlight_compensation 0x0098091c (int)    : min=0 max=1 step=1 default=57343 value=1

Camera Controls

                  auto_exposure 0x009a0901 (menu)   : min=0 max=3 default=0 value=3 (Aperture Priority Mode)
                                1: Manual Mode
                                3: Aperture Priority Mode
         exposure_time_absolute 0x009a0902 (int)    : min=3 max=2047 step=1 default=250 value=156 flags=inactive
     exposure_dynamic_framerate 0x009a0903 (bool)   : default=0 value=1
                   pan_absolute 0x009a0908 (int)    : min=-36000 max=36000 step=3600 default=0 value=0
                  tilt_absolute 0x009a0909 (int)    : min=-36000 max=36000 step=3600 default=0 value=18000
                 focus_absolute 0x009a090a (int)    : min=0 max=255 step=5 default=8189 value=50 flags=inactive
     focus_automatic_continuous 0x009a090c (bool)   : default=1 value=1
                  zoom_absolute 0x009a090d (int)    : min=100 max=500 step=1 default=57343 value=200
```

TzuHuanTai, 2025-03-01:
Because it's a WHEP source, you need a whep player, such as [@eyevinn/webrtc-player](https://www.npmjs.com/package/@eyevinn/webrtc-player) or so, to play the steaming.
Click their online demo, then type the `http://127.0.0.1:8080` in the URL field and play it.

closed 2025-04-17 by TzuHuanTai