mazupo/pi-webrtc #253: Unable to view video when iphone is using 5G

iragreenstein, 2025-03-17:
First off, absolutely great app chain!

Chain:

- Pi 5 to host the pi_webrtc application
- HiveMQ as the cloud MQTT server
- iphone connected to https://picamera.live for viewing

When my iphone is using wifi, I can view the video (and also have audio) while using the Safari viewer.

When I switch off wifi on the iphone, i am on 5G. The UUID nor the network settings have not changed, but the dot in preview view turns red.

Do I need to inject a STUN server to make 5G work ?

Thanks!

TzuHuanTai, 2025-03-17:
Yes, STUN is required for 5G, see [here](https://developer.mozilla.org/en-US/docs/Web/API/WebRTC_API/Protocols#stun).
TURN server is also required If your carrier use symetric NAT.

iragreenstein, 2025-03-17:
Thank you for the quick answer!

I was trying to include a TURN server (--turn_url turn:openrelay.metered.ca:3478?transport=tcp), but I would receive the error: Turn url should start with "turn:"

I see it is coming from parser.cpp, lines 119 - 122.

I think:
`if (!args.turn_url.empty() && args.turn_url.substr(0, 4) == "turn")`

should be:
`if (!args.turn_url.empty() && args.turn_url.substr(0, 4) != "turn")`

Thanks!

TzuHuanTai, 2025-03-17:
Oops... I didn't find there is a bug.
Can you add the TURN in the network setting from the web client side before I release the new binary?
Turn can be used as long as either side has set it up.

closed 2025-04-09 by TzuHuanTai