mazupo/pi-webrtc #264: [main] Error: No signaling service is running.

b4iterdev, 2025-04-16:
I'm having this error when trying to start `pi_webrtc` with WHEP:
```
./pi_webrtc --fps=30 --width=1280 --height=720 --http_port=8080 --uid=test  --hw_accel
```

and the output:
```
Using Libcamera, ID: 0
[0:18:43.785126742] [900]  INFO Camera camera_manager.cpp:327 libcamera v0.4.0+53-29156679
[0:18:43.875247843] [913]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:18:43.881019526] [913]  INFO RPI vc4.cpp:447 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media3 and ISP device /dev/media0
[libcamera_capturer] camera id: /base/soc/i2c0mux/i2c@1/imx219@10
[libcamera_capturer] camera validated format: 1280x720-YUV420.
[libcamera_capturer]   width: 1280, height: 720, stride: 1280
[0:18:43.882683392] [900]  INFO Camera camera.cpp:1202 configuring streams: (0) 1280x720-YUV420
[0:18:43.883432749] [913]  INFO RPI vc4.cpp:622 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1920x1080-SBGGR10_1X10 - Selected unicam format: 1920x1080-pBAA
[main] Error: No signaling service is running.
```

I have updated the OS to the latest version using `sudo apt update` and `sudo apt full-upgrade`, also I installed `libmosquitto1 pulseaudio libavformat59 libswscale6 nginx` as indicated in instruction, is there anything I'm missing ?

TzuHuanTai, 2025-04-16:
you miss the `--use_whep` flag. [example](https://github.com/TzuHuanTai/RaspberryPi-WebRTC/wiki#whep)

b4iterdev, 2025-04-16:
Oh yeah I have only read at https://github.com/TzuHuanTai/RaspberryPi-WebRTC/wiki/Advanced-Settings#whep-with-nginx-proxy, which does not have that flag lol

closed 2025-04-16 by b4iterdev

TzuHuanTai, 2025-04-16:
sry, my bad. I added the flag. lol