mazupo/pi-webrtc #267: suggestion: add awb and/or tuning file support for libcamera

b4iterdev, 2025-05-01:
This is particularly useful for user who use camera sensor without an infrared filter (NoIR) on a infrared filter version, which require either to use `noir.json` tuning file or pass `-awb greyworld` as a workaround.

Reference: https://forums.raspberrypi.com/viewtopic.php?t=347194
https://www.waveshare.com/wiki/Raspberry_Pi_HQ_Camera#rpicam-hello

TzuHuanTai, 2025-05-02:
[image]

b4iterdev, 2025-05-02:
I don't use MQTT, but use WHEP instead. Futhermore I'd prefer to have a built-in option rather than having ability to adjust it in player

TzuHuanTai, 2025-05-02:
I see. It indeed a useful option for WHEP users. The feature may be added in the next version.

closed 2025-05-08 by TzuHuanTai