mazupo/pi-webrtc #268: datachannel supports

jcl2023, 2025-05-01:
Does the app code support datachannel?
If yes, how do I use it to send command to PI via a web browser?
Is there a doc to describe this?

Thanks,

TzuHuanTai, 2025-05-02:
I only designed send commands to get a snapshot or mp4 file now, see the [example](https://github.com/TzuHuanTai/picamera.js?tab=readme-ov-file#example)

jcl2023, 2025-05-02:
I just want to send a "hello" message to server on pi. 
Is something like below I should use?

conn.onDatachannel = (dc) => {
  // on connected to remote datachannel
  conn.sendmsg("hello");
}

TzuHuanTai, 2025-05-02:
This is not available now.
Actually it's not difficult to implement one, and I've thought about the simular feature, sending the massage to server and broadcast the message to local socket port that I can control i2c/spi/gpio via IPC. I haven't make the final decision though, because I found some people already use MQTT to do the same things, it might be a duplicated feature.

jcl2023, 2025-05-02:
My  usage will be  to control GPIO.
The main benefit of using datachannel instead of mqtt is latency (I guess).
This feature is quite useful. Please consider to add it

Thanks,

jcl2023, 2025-05-03:
Does picamera.js provide a send emssage API so I can use it on web client?
If not, how can I use picamera.js to send a generic message (string)?

Thanks,

TzuHuanTai, 2025-05-04:
I don't implement any way to send a string message now. I'm doing a broadcast function this moment. I can add this datachannel feature after I finish the things on my hand.

TzuHuanTai, 2025-05-09:
Could you briefly describe your use case for me? I'd like to broadcast the message from datachannel to local socket port that enable other programs can read it via subscribe the same port. But I don't know if this a good approach for you?

TzuHuanTai, 2025-05-09:
"broadcast" means sending the same message to multiple local port.
e.g.  `./pi-webrtc .... --ipc-port=1001,1002`
That allow other two programs listen the same msg from the different port.

jcl2023, 2025-05-10:
One more thing, how picamera.js handles message sent from pi-webrtc? This is also useful for libdatachannel communications

Thanks,

closed 2025-05-15 by TzuHuanTai

TzuHuanTai, 2025-05-15:
picamera.js@v1.1.0 now supports a new option: ipcMode. You can check the usage example [here](https://www.npmjs.com/package/picamera.js#send-message-for-ipc-via-datachannel).
Please update to the latest `pi-webrtc` `v1.1.0` and run it with the `--enable-ipc` flag, see [doc](https://github.com/TzuHuanTai/RaspberryPi-WebRTC/wiki/Advanced-Settings#two-way-datachannel-messaging).

I’ve tested all the functions and they are working well.
Hope this new feature helps your use case! :)

jcl2023, 2025-05-15:
Thanks to enable this feature and I will let you know how it works after I try it

jcl2023, 2025-05-15:
I tried to modify your index,html to add send message function and host this index.html on my own http server (python3 -m http.server 9000)
I got this error message below when I click start button. If I use index.html hosted on your webserver, I don't see this error message.

Uncaught TypeError: crypto.randomUUID is not a function
    at new Tr (picamera.esm.js:5:22867)
    at new Rn (picamera.esm.js:9:9424)
    at start (index.html:62:14)
    at HTMLButtonElement.onclick (index.html:25:33)

Thanks,

TzuHuanTai, 2025-05-15:
Crypto.randomUUID() is only avaliable in https.
https://developer.mozilla.org/en-US/docs/Web/API/Crypto/randomUUID

jcl2023, 2025-05-15:
I don't have a https server yet. Can you add a sample webpage for me to test bidirectional data communications?

Thanks,