mazupo/pi-webrtc #269: segmentation fault with hw_accel enabled on pi zero 2w

jcl2023, 2025-05-02:
When I run the following script, I got segmentation fault:

./run_webrtc.sh: line 15:  5266 Segmentation fault      ./pi_webrtc --camera=v4l2:0 --v4l2_format=mjpeg --fps=30 --width=1024 --height=768 --use_mqtt --mqtt_host=2c88e4d4e0f948b98b6f5aab8c5a6f5b.s1.eu.hivemq.cloud --mqtt_port=8883 --mqtt_username=xxxx --mqtt_password=yyyy --uid=2540f45f-7011-499c-bb29-86c40b6d2635 --no_audio --hw_accel

If I remove --hw-accel, it runs OK and I can see video on web client.
Any idea why segmentation fault happens>

Thanks,

TzuHuanTai, 2025-05-02:
I guess the frame size is missing a mapping in rpi built-in hardware encoder. The openh264 software encoder is used if you remove `--hw-accel`, it support wider resolutions.

It'll be ok If you try 1280x720. I tested 1920x1088 will crash as well, see the caution info [here](https://github.com/TzuHuanTai/RaspberryPi-WebRTC/wiki/Advanced-Settings#using-the-legacy-v4l2-driver).

TzuHuanTai, 2025-05-02:
I tested 1024x768 is work on my pi zero 2w by using `--camera=libcamera:0`. 
Could you please run `v4l2-ctl -d /dev/video0 --list-formats`, and what the output is?

jcl2023, 2025-05-02:
My camera is usb webcam. The formats it supports are below:

 v4l2-ctl -d /dev/video0 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'MJPG' (Motion-JPEG, compressed)
                Size: Discrete 1920x1080
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 1280x1024
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 1280x960
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 1280x720
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 1024x768
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 960x720
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 800x600
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 640x360
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 1920x1080
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.033s (30.000 fps)
        [1]: 'YUYV' (YUYV 4:2:2)
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 1920x1080
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 1280x720
                        Interval: Discrete 0.100s (10.000 fps)
                Size: Discrete 1024x768
                        Interval: Discrete 0.100s (10.000 fps)
                Size: Discrete 800x600
                        Interval: Discrete 0.050s (20.000 fps)
                Size: Discrete 800x448
                        Interval: Discrete 0.100s (10.000 fps)
                Size: Discrete 720x544
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.033s (30.000 fps)

closed 2025-05-19 by TzuHuanTai