mazupo/pi-webrtc #280: Problem with RP camera module 3?

dirkvdb2, 2025-05-12:
On a raspberry pi 4 with raspberry camera module 3:

./pi-webrtc \
    --camera=libcamera:0 \
    --uid=home-pi-4b \
    --fps=30 \
    --width=1280 \
    --height=960 \
    --use-whep \
    --no-audio \
    --http-port=8080

Runs, but browsers (chromium and firefox; http://localhost:8080) say: 'Without content type'

libcamera 0 verfied, works with libcamera-hello and

libcamera-vid -t 5000 --width 1280 --height 960 --framerate 30 -o test.h264

Available cameras
-----------------
0 : imx708 [4608x2592 10-bit] (/base/soc/i2c0mux/i2c@1/imx708@1a)
    Modes: 'SBGGR10_CSI2P' : 1536x864 [30.00 fps - (65535, 65535)/65535x65535 crop]
                             2304x1296 [30.00 fps - (65535, 65535)/65535x65535 crop]
                             4608x2592 [30.00 fps - (65535, 65535)/65535x65535 crop]

tried with: ; same result

./pi-webrtc \
    --camera=libcamera:0 \
    --uid=home-pi-4b \
    --fps=30 \
    --width=1536 \
    --height=1296 \
    --use-whep \
    --no-audio \
    --http-port=8080

dirkvdb2, 2025-05-12:
Ah, forgot to add terminal output:
dirk@raspberrypi:~ $ ./pi-webrtc \
    --camera=libcamera:0 \
    --uid=home-pi-4b \
    --fps=30 \
    --width=1536 \
    --height=1296 \
    --use-whep \
    --no-audio \
    --http-port=8080
Using Libcamera, ID: 0
[2:00:48.089232294] [6797]  INFO Camera camera_manager.cpp:326 libcamera v0.5.0+59-d83ff0a4
[2:00:48.145337692] [6805]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[2:00:48.148429814] [6805]  INFO RPI vc4.cpp:447 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media1 and ISP device /dev/media2
[libcamera_capturer] camera id: /base/soc/i2c0mux/i2c@1/imx708@1a
[libcamera_capturer] camera validated format: 1536x1296-YUV420.
[libcamera_capturer]   width: 1536, height: 1296, stride: 1536
[2:00:48.224090376] [6797]  INFO Camera camera.cpp:1205 configuring streams: (0) 1536x1296-YUV420
[2:00:48.224546795] [6805]  INFO RPI vc4.cpp:622 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected unicam format: 2304x1296-pBAA
[http_service] Http server is running on http://*:8080
^C
dirk@raspberrypi:~ $

TzuHuanTai, 2025-05-12:
It's WHEP source, not a general http web, so you need to use a [whep player](https://www.npmjs.com/package/@eyevinn/webrtc-player) to watch it. The online demo runs on **https**. Set up your Pi to serve https using the WHEP [guide](https://github.com/TzuHuanTai/RaspberryPi-WebRTC/wiki/Advanced-Settings#whep-with-nginx-proxy) with Nginx, then try the URL.
[image]

closed 2025-06-02 by TzuHuanTai