mazupo/pi-webrtc #285: Is the STUN/TURN functionality effective for pi-webrtc streaming over 5G?

yjshappy, 2025-05-30:
Thank you very much for open-sourcing this project!

I used it on a Raspberry Pi 4B for streaming. First, in the local network, I successfully achieved real-time playback of the USB camera's video on pi camera with a latency of around 200ms, which meets my needs.

Then, I connected the Raspberry Pi to a 5G router and deployed coturn on Tencent Cloud as a STUN/TURN server to assist WebRTC in NAT traversal. However, I am now unable to view the video on pi camera.

My configuration on pi camera is as follows:

[image]

First, I can confirm that my coturn server is operational and working correctly, as I have successfully tested it using Trickle ICE.

[image]

Then i also tested ：

turnutils_uclient -v -t -T -u admin -w 123456 12.***.***.** 

0: IPv4. Connected from: 172.16.10.38:41044
0: IPv4. Connected to: 12.***.***.**:3478
0: allocate sent
0: allocate response received: 
0: allocate sent
0: allocate response received: 
0: success
0: IPv4. Received relay addr: 12.***.***.**:55621
0: clnet_allocate: rtv=0
0: refresh sent
0: refresh response received: 
0: success
0: IPv4. Connected from: 172.16.10.38:41050
0: IPv4. Connected to: 12.***.***.**:3478
0: IPv4. Connected from: 172.16.10.38:41064
0: IPv4. Connected to: 12.***.***.**:3478
0: allocate sent
0: allocate response received: 
0: allocate sent
0: allocate response received: 
0: success
0: IPv4. Received relay addr: 12.***.***.**:51967
0: clnet_allocate: rtv=0
0: refresh sent
0: refresh response received: 
0: success
0: allocate sent
0: allocate response received: 
0: allocate sent
0: allocate response received: 
0: success
0: IPv4. Received relay addr: 12.***.***.**:54402
0: clnet_allocate: rtv=0
0: refresh sent
0: refresh response received: 
0: success
0: create perm sent: 12.***.***.**:54402
0: cp response received: 
0: success
0: create perm sent: 12.***.***.**:51967
0: cp response received: 
0: success
0: tcp connect sent
0: connection bind sent
0: connect bind response received: 
0: success
0: IPv4. TCP data network connected to: 12.***.***.**:3478
0: connection bind sent
0: connect bind response received: 
0: success
0: IPv4. TCP data network connected to: 12.***.***.**:3478
1: Total connect time is 1
1: 2 connections are completed
1: start_mclient: msz=2, tot_send_msgs=0, tot_recv_msgs=0, tot_send_bytes ~ 0, tot_recv_bytes ~ 0
2: start_mclient: msz=2, tot_send_msgs=0, tot_recv_msgs=0, tot_send_bytes ~ 0, tot_recv_bytes ~ 0
3: start_mclient: msz=2, tot_send_msgs=5, tot_recv_msgs=5, tot_send_bytes ~ 500, tot_recv_bytes ~ 500
4: start_mclient: msz=2, tot_send_msgs=5, tot_recv_msgs=5, tot_send_bytes ~ 500, tot_recv_bytes ~ 500
5: start_mclient: msz=2, tot_send_msgs=5, tot_recv_msgs=5, tot_send_bytes ~ 500, tot_recv_bytes ~ 500
6: done, connection 0x7fe8e0fb2010 closed.
6: done, connection 0x7fe8e08f5010 closed.
6: start_mclient: tot_send_msgs=10, tot_recv_msgs=10
6: start_mclient: tot_send_bytes ~ 1000, tot_recv_bytes ~ 1000
6: Total transmit time is 5
6: Total lost packets 0 (0.000000%), total send dropped 0 (0.000000%)
6: Average round trip delay 6.700000 ms; min = 4 ms, max = 9 ms
6: Average jitter 0.800000 ms; min = 0 ms, max = 3 ms

When viewing the video through pi camera, the WebRTC server outputs the following log information:

/work/RaspberryPi-WebRTC/build$ ./pi-webrtc   --camera=v4l2:1   --uid=camera_0   --v4l2-format=mjpeg --fps=30     --width=640     
--height=480 --mqtt-host=44b78df5e3fa4c51bfac*********.hivemq.cloud   --mqtt-port=8883   --mqtt-username=*****   --mqtt-password=*****  
 --use-mqtt --no-adaptive
Using V4L2, ID: 1
Using V4L2, format: mjpeg
[conductor] network thread start: success!
[conductor] worker thread start: success!
[conductor] signaling thread start: success!
[v4l2_utils] Open file /dev/video1 fd(26) success!
[v4l2_utils] fd(26) driver 'uvcvideo' on card 'USB 2.0 Camera' in splane mode
[v4l2_capturer]   Fps: 30
[v4l2_capturer]   Rotation: 0
[v4l2_utils] Error: fd(26) set ctrl(9963810): Invalid argument
[v4l2_utils] fd(26) original formats: MJPG(640x480)
[v4l2_utils] fd(26) latest format: MJPG(640x480)
[v4l2_utils] Error: fd(26) set ext ctrl(10029519): Invalid argument
[v4l2_utils] fd(26) query buffer at 0xffffadd35000 (length: 307200)
[v4l2_utils] fd(26) query buffer at 0xffffadcea000 (length: 307200)
[v4l2_utils] fd(26) query buffer at 0xffffadc9f000 (length: 307200)
[v4l2_utils] fd(26) query buffer at 0xffffadc54000 (length: 307200)
[main] Recorder is not started!
[v4l2_capturer] capture timeout
[v4l2_capturer] capture timeout
[v4l2_capturer] capture timeout
[mqtt_service] Successfully subscribed to topic: camera_0/sdp/+/offer
[mqtt_service] Successfully subscribed to topic: camera_0/ice/+/offer
[mqtt_service] MQTT service is ready.
[mqtt_service] Received remote ICE: 0, 0, candidate:2257720285 1 udp 2122260224 172.16.10.38 60445 typ host generation 0 ufrag QEz2 network-id 1
[mqtt_service] Received remote ICE: 0, 0, candidate:997987654 1 udp 2122194688 172.16.10.86 54940 typ host generation 0 ufrag QEz2 network-id 2 network-cost 10
[mqtt_service] Received remote ICE: 0, 0, candidate:1692052767 1 udp 25108224 1.12.**** 59556 typ relay raddr 58.62.103.185 rport 5811 generation 0 ufrag QEz2 network-id 1
[mqtt_service] Received remote ICE: 0, 0, candidate:4161550032 1 tcp 1518280448 172.16.10.38 9 typ host tcptype active generation 0 ufrag QEz2 network-id 1
[mqtt_service] Received remote ICE: 0, 0, candidate:1172451403 1 tcp 1518214912 172.16.10.86 9 typ host tcptype active generation 0 ufrag QEz2 network-id 2 network-cost 10
[rtc_peer] The Command data channel is established successfully.
[conductor] Peer connection(d5114432-3f27-4b4a-b8da-e6056676cb4d) is created! 
[mqtt_service] Received remote [offer] SDP: v=0
o=- 2190910677192188153 2 IN IP4 127.0.0.1
s=-
t=0 0
a=group:BUNDLE 0
a=extmap-allow-mixed
a=msid-semantic: WMS
m=application 9 UDP/DTLS/SCTP webrtc-datachannel
c=IN IP4 0.0.0.0
a=ice-ufrag:QEz2
a=ice-pwd:BGe3tibnlJBF/dZyImgjnGD2
a=ice-options:trickle
a=fingerprint:sha-256 C3:F2:37:42:5A:FF:F1:8B:59:4E:86:60:C7:7C:46:D6:FB:79:17:A3:22:46:D8:20:FA:70:2B:CF:AC:8C:68:02
a=setup:actpass
a=mid:0
a=sctp-port:5000
a=max-message-size:262144

[rtc_peer] OnSignalingChange => have-remote-offer
[rtc_peer] => Set sdp success!
[rtc_peer] OnSignalingChange => stable
[mqtt_service] Answer local [answer] SDP: v=0
o=- 7364830024270440023 2 IN IP4 127.0.0.1
s=-
t=0 0
a=group:BUNDLE 0
a=extmap-allow-mixed
a=msid-semantic: WMS
m=application 9 UDP/DTLS/SCTP webrtc-datachannel
c=IN IP4 0.0.0.0
a=ice-ufrag:baM7
a=ice-pwd:0BBc/CWCRnekqBmE5g6adWpH
a=ice-options:trickle
a=fingerprint:sha-256 D9:10:7E:79:1B:57:9F:F5:35:59:56:96:88:EC:CC:52:33:CF:33:69:B5:AF:CE:CE:0D:58:34:96:8E:14:29:B5
a=setup:active
a=mid:0
a=sctp-port:5000
a=max-message-size:262144

[rtc_peer] => Set sdp success!
[rtc_peer] OnIceGatheringChange => gathering
[mqtt_service] Sent local ICE:  0, 0, candidate:2584966873 1 udp 2122194687 192.168.0.154 36766 typ host generation 0 ufrag baM7 network-id 1
[mqtt_service] Sent local ICE:  0, 0, candidate:2078630704 1 udp 2122262783 2408:8956:5020:6b37:f527:9469:2f48:d324 51380 typ host generation 0 ufrag baM7 network-id 2
[mqtt_service] Sent local ICE:  0, 0, candidate:3839661121 1 tcp 1518214911 192.168.0.154 49815 typ host tcptype passive generation 0 ufrag baM7 network-id 1
[mqtt_service] Sent local ICE:  0, 0, candidate:86671784 1 tcp 1518283007 2408:8956:5020:6b37:f527:9469:2f48:d324 34881 typ host tcptype passive generation 0 ufrag baM7 network-id 2
[mqtt_service] Sent local ICE:  0, 0, candidate:2601714943 1 udp 1685987071 112.96.211.94 31477 typ srflx raddr 192.168.0.154 rport 36766 generation 0 ufrag baM7 network-id 1
[rtc_peer] OnIceGatheringChange => complete
[rtc_peer] Connection timeout after kConnecting. Closing connection.
[rtc_peer] OnConnectionChange => closed
[rtc_peer] OnSignalingChange => closed
[mqtt_service] Found peer_id key: d5114432-3f27-4b4a-b8da-e6056676cb4d, connected value: 0
[mqtt_service] (d5114432-3f27-4b4a-b8da-e6056676cb4d) was erased.
[rtc_peer] peer connection (d5114432-3f27-4b4a-b8da-e6056676cb4d) was destroyed!
[data_channel_subject] datachannel (command) is released!

yjshappy, 2025-05-31:
Thank you very much for your reply!

So, you mean that if the Raspberry Pi connects to a 5G network at home and then connects to a local 5G network elsewhere, the Pi Camera's web access will be available without needing to set up TURN, is that correct? Could it be that you're using a special 5G network that has already implemented internal NAT traversal? @TzuHuanTai

TzuHuanTai, 2025-05-31:
My raspberry pi connects to the home wifi, and my phones use normal 5G mobile data when I am out of my place. I didn't do anything settings about it, just choose the cheapest plan from T-Mobile and AT&T, then watch it without TURN.

closed 2025-06-06 by yjshappy