mazupo/pi-webrtc #295: pi-camera API calls needed

Harvey-Mushman, 2025-07-05:
Once pi-Camera is installed and running perfectly, in my app I need to find a way to record video on the RPi under my control.  I have the pi-Camera recording function working to an external USB drive, that would as described but it does not give me the ability to start/stop a recording and then save the video into a filename I designate.  

I thought I might be able to preform this function on the RPi by opening a local browser but the USB mic is currently single user and Chrome can't attach to it.  Same is true for the video stream, Chrome won't let me have access to it.

So I'm wondering if adding function calls to the active pi-camera that allow me to start/stop video and audio recording on the local RPi can be added in a future release?  My intention would be when an event occurs on the GPIO bus of the RPi, I would start a timestamped recording with the event included in the filename.

Thanks in advance for your consideration.  BTW - I love your work!

TzuHuanTai, 2025-07-05:
I agree It's useful but I need to get things done first on my hands.🤣

Harvey-Mushman, 2025-07-05:
Understood, but I also just discovered, with a usb microphone the recording that runs continuously does not include the audio track, is this a bug or a feature ?  I tested with audio music playing in the background and in the video I started pointing to my loud speaker.  I then found the mp3 recording on my external disk and played it in VLC.  No audio was output.

My thinking for a workaround was to timestamp the start and stop times into a log and then manually if needed go back and find the recordings but without the audio it does me no good.

closed 2025-08-21 by TzuHuanTai